PIC-IRIS/PH5 #4: Creating a test PH5 Dataset

nick-iris, 2016-11-09:
@derick-hess @rsdeazevedo 
I would like to create a very small test PH5 dataset that I can use for unit testing. Do you know what files I would have to modify to create a as minimal as possible PH5 dataset that can be tracked in Git?

derick-hess, 2016-11-09:
the easiest way to do it is to get some raw rt130 and or rt125a data and build a ph5 experiment.

Here is a document we have on creating a ph5 experiment http://crack.seismo.unr.edu/ftp/pub/louie/class/492/data/2016/deepremi/PH5_inst/PH5Nut.pdf

nick-iris, 2016-11-09:
Okay thanks. Do have some raw rt130 or rt125a data that I can use? I need to make the PH5 experiment as small as possible.

derick-hess, 2016-11-09:
Yeah I'll get you some rt125 data and the csv files form seisorz then you can just take a few of them and modify the csv files... you are still looking at 90MB for a single rt125 file that was only out a few days though

derick-hess, 2016-11-09:
Oh I have talked ot steve about this but I plan on making an extensive ph5 test experiment. It will probably be  multiple GB but its going to have data from a bunch of different instruments. KNown good and bad data, odd date, etc to test the PH5 software against to make sure it is handling everything

nick-iris, 2016-11-09:
Yeah 90mb would be too big to store in Git. Is there a way to create one that is under 1mb? The data can be junk, since it is just for testing.

Creating a experiment with various known edge cases sounds like a good idea. You will just need to make sure that your code still runs if the test experiment does not exist, since you will not want to track the file in GitHub. I would suggest writing the tests in a generic way, so that they work with any arbitrary PH5 file. This way we could potentially track a extremely small default experiment with the source code that the test will use by default, but also run test on the more extensive ph5 test experiment when one is present.

derick-hess, 2016-11-09:
It might be possible putting basically no data in the PH5 file but not sure. Even without and data or receiver/event metadata the master.ph5 alone before it has all it's tables is close to 1MB

nick-iris, 2016-11-09:
What do you think the maximum practical size is for a file tracked in Git? Would be be able to make a PH5 file under 10mb? I think there is an advantage to having a test data tracked with the source code so that you can run test independently of the 2GB test data set that cannot practically be tracked in Git.

derick-hess, 2016-11-09:
I think under 10MB is doable.

nick-iris, 2016-11-09:
I discussed this a little more with Adam, and he suggested that we write ad-hoc tests for the PH5 APIs. This would imply not keeping any test PH5 data with the code, but instead making the test a separate internal way of confirming that things are working correctly before pushing changes. 

This matches the plan you proposed, so writing test that require a 2GB internal dataset shouldn't be a problem. However, I still think that any additional tests that don't require the large test dataset should be tracked on GitHub and kept in a separate test file along with the API source code (e.g. webservices/tests/test_ph5tomsapi.py).

nick-iris, 2017-03-22:
I'm closing this issue. I've since wrote tests for the web services that use experiments in the archive instead.

closed 2017-03-22 by nick-iris