PIC-IRIS/PH5 #5: Validating PH5 arguments

nick-iris, 2016-11-10:
I'm working on adding better validation checks for web service query arguments. Here are some questions regarding certain PH5 query arguments that I have:

- Will report numbers always contain a two digit first number followed by a dash followed by a three digit number? (e.g. 15-021)
- Will a array-id always be a three digit numeric value? (e.g. 001)
- Will a shot-id always be four digit numeric value? (e.g. 1001)
- What is the format for receiver ids?
- Does the "length" parameter need to be a non-negative integer? (e.g. 0, 1, 2, etc..)
- Does the "offset" parameter need to be a non-negative integer?
- Does the "component" parameter need to be a non-negative integer? Is there a restriction on the number of digits like with the SEED channel code?
- Does the "reduction velocity" parameter need to be a non-negative integer?
- Can the extended header parameter only be "P", "S", or "U"?
- What would some example minlat, maxlat, minlon, and maxlat values look like in the ph5tostationxml.py service?
- What does a example "shotline" argument value look like?

Thanks

derick-hess, 2016-11-10:
- report number should always be in that format. Mea or whoever at the DMC assigns them can answer for sure.
- I believe array-id can in theory be any length. Practically though 3 digits is way more than enough
- shot-id, no it can be any length. ENAM for example is a 7 digit number. Offshore experiments it is possible that you have 100,000+ shots
- receiver-id is generally a 5 digit non negative integer but there are experiments where it is just an integer with count starting at 1
- length has to be a non negative number. It doesn't have to be an integer.
- offset can be a negative number it also doesn't have to be an integer
- component is a non negative number. Theoretically there is no limit to the number of components
- minlat, maxlat, minlon, maxlon follow the fdsn specifications
- shotline acts exactly like an array_id with the exception that you can only specify a single shotline per request instead of a comma separated list.

nick-iris, 2016-11-11:
Thanks Derick. Are array-id, shot-id, receiver-id, component, and shotline *always* numeric, or can they sometimes contain letters too?

derick-hess, 2016-11-11:
yes, they should all always be numeric.

nick-iris, 2016-11-14:
Will the shot-id, array-id, receiver-id, and component-id list arguments accept "*" and "?" wildcards? I currently validate that they are positive integers, but do not accept wildcards.

nick-iris, 2016-11-14:
I've attached a library for parsing web service input arguments, as well as, related unit tests. As I discussed with Derick earlier today, it would be good to use these methods for validating/parsing inputs for both the PH5 Web Service and PH5 APIs. This way everything is consistent.

Let me know if you see anything that should be changed. I will also add these files [here](https://github.com/PIC-IRIS/PH5/tree/master/webservices) if that is better for the long term tracking of this code. Just let me know. Thanks.

[Arguments Parser.zip](https://github.com/PIC-IRIS/PH5/files/590630/Arguments.Parser.zip)

nick-iris, 2016-11-16:
Adam brought to my attention that these methods are really no substitute for actually parsing and using the users input arguments in the APIs. Parsing and using arguments in the API is the ultimate judge of whether or not a input is valid. There would be no way to guarantee that my checks would catch every invalid or valid input.

I will run these methods, with some looser constraints, in the PH5 Web Service to eliminate any obviously wrong inputs before passing off the values to the APIs. You should continue to run your own checks when you parse the inputs in the APIs. Sorry for the confusion.

closed 2017-01-04 by nick-iris