PIC-IRIS/PH5 #6: ph5tostationxml.py missing "text" format

nick-iris, 2016-11-16:
@derick-hess @rsdeazevedo
It is critical that we add the ability to return StationXML from ph5tostationxml.py in "text" format, in addition to the "xml" format that is currently supported. What is the current status of this feature?

derick-hess, 2016-11-16:
I gave that to one of our students as a project. I'll check with him to see how it's going. If it seems it will take too long I can do it pretty quickly.

nick-iris, 2016-12-01:
@derick-hess  
It might be a good idea to structure this code so that it could be added to ObsPy at a later date. You could then just call the Inventory "write" method with the format set to TEXT:
ie. `inv.write(sys.stdout, format='TEXT', nsmap = {'iris': iris_custom_ns})`

nick-iris, 2016-12-02:
@derick-hess @rsdeazevedo  It appears that the ObsPy developers have already started developing code for writing StationTXT. See ObsPy issue 1466 https://github.com/obspy/obspy/pull/1466/:

https://github.com/obspy/obspy/pull/1466/commits/01b4df4318c892fbd613028e558087268051c732

I will contact the ObsPy developers and work on resolving any remaining issues and getting this officially supported in ObsPy. Until then, I have copied their method for writing an Inventory as StationTXT to the web handler and will add this later today. You probably should tell your intern not to worry about this.

nick-iris, 2016-12-21:
I just finished getting StationTXT added to ObsPy. It is scheduled to be in the ObsPy 1.1.0 release, so I'm going to close this issue.

closed 2016-12-21 by nick-iris