PIC-IRIS/PH5 #9: ph5tomsAPI not handling wildcards correctly

nick-iris, 2016-11-23:
@derick-hess @rsdeazevedo The ph5tomsAPI.py program no longer handles wildcards correctly for a number of cases. Support for wildcards is still required for request by shot to work.

A few examples of requests that return no data when they should are:
`python ph5tomsAPI.py -n master.ph5 -p /hdf5-data2/PH5_Experiments/pn4/13-005/ --stream --shotline 001 --eventnumbers 5003 --array 001 --length 60 --station 100?`

`python ph5tomsAPI.py -n master.ph5 -p /hdf5-data2/PH5_Experiments/pn4/13-005/ --stream --shotline 001 --eventnumbers 5003 --array 001 --length 60 --station *`

`python ph5tomsAPI.py -n master.ph5 -p /hdf5-data2/PH5_Experiments/pn4/13-005/ --stream --shotline 001 --eventnumbers 5003 --array 001 --length 60 --channel EP?`

`python ph5tomsAPI.py -n master.ph5 -p /hdf5-data2/PH5_Experiments/pn4/13-005/ --stream --station=* --starttime 2014-11-23T10:00:00.0 --stoptime 2014-11-23T15:00:00.0`

This one actually errors out.
`python ph5tomsAPI.py -n master.ph5 -p /hdf5-data2/PH5_Experiments/pn4/13-005/ --stream --shotline 001 --eventnumbers 5003 --array=* --length 60`

derick-hess, 2016-11-23:
I'll have this fixed  soon. Working on it now.

nick-iris, 2016-11-23:
Okay thanks. I think shotline and shotid should also accept wildcards.

Just FYI, I can expand most of the wildcards out in the web handler, but cannot handle shot-ids and shotlines since the ph5toexml.py program can currently only return _all_ event info for one network. Eventually I think we will want to change the ph5toexml.py program to accept additional query arguments. Then I could expand all wildcards in a list of requests in the handler code and remove overlapping/duplicate requests, in order to add support for HTTP POST for the by Shot request type. This is what I do currently for the FDSN request type.

derick-hess, 2016-11-23:
shotid will be easy to accept wildcards, shotline I'll have to work on a bit since you can only have a maximum of one shotline per request and wildcards could match multiple shotlines.

nick-iris, 2016-11-23:
In that case maybe we only support wildcards for shotid and not shotline. Do you think supporting wildcards for shotlines would be a useful feature?

derick-hess, 2016-11-23:
What I'm going to do now is push code that fixes the array, station, and channel wildcards sicne that is done.

I think we should definitely support wildcards for shotlines and shotids as well as requesting  multiple shotlines. I can fix this but it will take substantially more coding and probably more time than I have today to fix those wildcards and add the the ability for multiple shotlines. I will work on this and try to get that pushed by the end of the weekend.

nick-iris, 2016-11-23:
Okay that sounds good. Don't worry about working on it over the holiday weekend unless you want to. The FDSN service still works fine since I expand wildcards in the web handler as I described above. I would say that proper error handling is equally, if not more, important as supporting wildcards for shotid and shotline.

nick-iris, 2016-11-23:
Can you add wildcard support for component-id easily to ph5tomsAPI.py?

derick-hess, 2016-11-23:
Yeah I can add that no problem.

nick-iris, 2017-01-05:
As long as we are updating ph5tomsAPI could we also change the eventnumbers parameter in the constructor to take a List instead of a comma separated String so that it matches the other input parameters?

i.e. `eventnumbers=""` would be changed to `eventnumbers=[]` here:
https://github.com/PIC-IRIS/PH5/blob/master/webservices/ph5tomsAPI.py#L99

derick-hess, 2017-01-05:
i'll work on fixing all of those.

nick-iris, 2017-02-13:
Thanks for pushing your fixes @derick-hess. All of the changes you made seem to be good, so I'm closing this issue.

closed 2017-02-13 by nick-iris