PIC-IRIS/PH5 #11: ph5tostationxml.py response level requests do not return response information

nick-iris, 2016-12-02:
Response level requests made to ph5tostationxml.py currently do not return any response level information. I believe this was because response info is being stored in PH5 as RESP files, and there is no easy way to read RESP files in Python.

What is the current status of this? Last I remember hearing was that a developer at the PIC had developed a library for this purpose, and was going to add it to ObsPy for the upcoming release.

derick-hess, 2016-12-02:
I'll talk to LLoyd tomorrow. I know it is ready to go for everything but nodes and that code is slated to be in the next Obspy release.  LLoyd was finishing up code to work for the nodes(or any other all in one unit where the sensor and datalogger don't have separate response information)

nick-iris, 2016-12-06:
Do you know what ObsPy GitHub issue LLoyd's code is under?

rsdeazevedo, 2016-12-06:
Lloyd;

Can you answer Nick?

Thanks.

nick-iris, 2016-12-06:
I found it. ObsPy issue 1185. Thanks.

dsentinel, 2016-12-07:
Glad you found it. If you have any questions feel free to ask. --Lloyd

nick-iris, 2017-05-08:
This is working with a developmental ObsPy 1.1.0 install, so I'm closing this issue.

closed 2017-05-08 by nick-iris