PIC-IRIS/PH5 #17: ph5tostationxml.py array argument does not filter results correctly

nick-iris, 2017-01-05:
I noticed that the `array` argument in ph5tostationxml.py does not filter results correctly. I believe one of the problems is with the ph5API module not returning all arrays for a given experiment. 

For example:
`python ph5tostationxml.py --nickname=master --ph5path=/hdf5-data2/PH5_Experiments/pn4/13-005/ --basepath=/hdf5-data2/PH5_Experiments/pn4/13-005/ --array=*`
returns a list of Array_t_names containing only the first array, even though array was set to *:
i.e. returns `self.ph5.Array_t_names = ['Array_t_001']`

nick-iris, 2017-01-05:
Perhaps this is a misunderstanding on my part. I've opened the Seisorz experiment in HDFView and see only one array listed under Sorts_g, so perhaps there is only Array_t_001 in this experiment? I think I was confused by the array_name_data_a field in the Receivers_g/Das_t tables. Is that something different? Thanks.

derick-hess, 2017-01-05:
I think Seisorz only has 1 array but I'll do some testing to make sure everythign is working correctly and close the issue this afternoon if things are ok.

nick-iris, 2017-01-05:
I think you're right. My bad, I shouldn't have jumped to this conclusion so quickly. It's probably safe to close this issue.

closed 2017-01-05 by derick-hess