PIC-IRIS/PH5 #21: Raise custom exceptions for known potential problems in PH5 APIs

nick-iris, 2017-03-15:
Runtime errors need to be caught and raise custom exceptions in the ph5tostationxml.py and ph5tomsAPI.py APIs.

You can create your own custom exceptions by subclassing the builtin Python Exception class.

In the Web Services code, I’ll then catch the known custom exception types that get raised by the PH5 APIs and then take an action that doesn’t break the services (i.e. skip the bad experiment and log the error.).

-------------
For example:

> Currently if one PH5 Experiment in the archive is missing a required piece of information, such as a SEED network code, it will break both of the Web Services. 

If a custom exception was raised in the APIs when the code tried to access the non-existent info, then I could have caught that exception and skipped+logged the bad experiment.

nick-iris, 2017-03-28:
All places where you are explicitly calling sys.exit() should instead be raising custom exceptions that I can catch in the Web Service. See the references below:

ph5tomsAPI.py
- https://github.com/PIC-IRIS/PH5/blob/master/webservices/ph5tomsAPI.py#L130-L136
- https://github.com/PIC-IRIS/PH5/blob/master/webservices/ph5tomsAPI.py#L143-L146
- https://github.com/PIC-IRIS/PH5/blob/master/webservices/ph5tomsAPI.py#L395-L396
- https://github.com/PIC-IRIS/PH5/blob/master/webservices/ph5tomsAPI.py#L398-L399
- https://github.com/PIC-IRIS/PH5/blob/master/webservices/ph5tomsAPI.py#L451-L454
- https://github.com/PIC-IRIS/PH5/blob/master/webservices/ph5tomsAPI.py#L831-L833

ph5tostationxml.py
- https://github.com/PIC-IRIS/PH5/blob/master/webservices/ph5tostationxml.py#L665-L667

nick-iris, 2017-03-28:
@derick-hess do you mind if I make a pull request with an initial crack at improving exception handling in ph5tomsAPI and ph5tostationxml.py? 

I'm pretty sure it will fix a bunch of compatibility issues with the Web Service.

For example, if a request by shot is made then the following code segment will kill the python process for any experiment that doesn't contain the requested shotline.

https://github.com/PIC-IRIS/PH5/blob/master/webservices/ph5tomsAPI.py#L395-L396

derick-hess, 2017-03-28:
sure go ahead

commit 15997cf9 (2017-03-29): Merge pull request #24 from PIC-IRIS/issue_21

Fix #21. Improve exception handling in ph5tomsAPI.py.
commit e6fff2fd (2017-03-29): Fix #21. Improve exception handling in ph5tomsAPI.py.
closed 2017-03-29 by derick-hess