PIC-IRIS/PH5 #33: ph5tostationxml.py queries against both PH5 Station-ID and SEED Station Code

nick-iris, 2017-04-24:
Currently the ph5tostationxml.py program compares values for the station argument against both the SEED Station Code and the PH5 Station-ID. 

This leads to misleading results. For example:
[http://service.iris.edu/ph5beta/station/1/query?network=4C,ZI&station=100?,DAN,IVY,LGC&format=text&level=channel](http://service.iris.edu/ph5beta/station/1/query?network=4C,ZI&station=100?,DAN,IVY,LGC&format=text&level=channel
)
returns all stations for 4C, when you expect to see only DAN, IVY, and LGC.

I think the logic should be changed to only compare against the SEED Station Code if one is defined. @derick-hess what do you think? Do all experiments have SEED Station Codes?

nick-iris, 2017-04-24:
If all experiments have SEED Station Codes defined then we should remove the following lines:
https://github.com/PIC-IRIS/PH5/blob/master/kitchen/ph5tostationxml.py#L518-L519

If some experiments only have PH5 station-ids set, we should only check against the station-id for cases where there is no SEED Station Code. However, ideally all experiments should have a defined SEED Station Code even if it is the same as the Station ID

i.e.
```
if not station_list[deployment][0] ['seed_station_name_s'] and \
    fnmatch.fnmatch(str(station), str(pattern)):
    # no seed station code defined so compare against ph5 station-id
    l.append(station)
elif fnmatch.fnmatch((station_list[deployment][0]
                        ['seed_station_name_s']), pattern):
    l.append(station)
```

derick-hess, 2017-04-25:
we can change that. In most cases they will be the same so just making it use seed station id for web service makes sense. I'll make those changes

nick-iris, 2017-04-25:
Why not just set both the SEED Station Code and PH5 Station-ID to the same value when they are the same, and then always use the SEED Station Code?

If this is too much of an undertaking, I think it should use the Station-ID only when a SEED Station Code is not defined. For future experiments I would set both values however, even if they are the same.

nick-iris, 2017-04-25:
> we can change that. In most cases they will be the same so just making it use seed station id for web service makes sense. I'll make those changes

Can you push this to a separate branch an make a pull request so I can also look it over? We'll want to get into the habit of avoiding pushing directly to master since we are nearing a production release.

commit 0efb012e (2017-04-25): Update ph5tostationxml.py to only check station-id if seed station codes are not defined. Fix #33
closed 2017-04-25 by nick-iris