PIC-IRIS/PH5 #47: Request by shot filename and actual event times mismatch

nick-iris, 2017-05-22:
I’ve noticed that the labeled dates/times on SAC files returned by for a “request by shot” largely differ from the event start times returned by the PH5 event web service

For example: 
http://service.iris.edu/ph5beta/dataselect/1/query?reqtype=shot&shotline=001&shotid=5011&array=001&length=60&reportnum=15-016&station=1002&format=sac&nodata=404

Returns a file labeled ZI.1002..DPZ.**2015.180.04.00.00**.SAC

http://service.iris.edu/ph5beta/event/1/query?catalog=15-016&format=shottext

The shot starttime for shotid=5011 however is **2015-06-29T09:33:19.000000Z**

2015.180.04.00.00 != 2015-06-29T09:33:19.0

We need to track-down and fix the cause of these time differences.

nick-iris, 2017-05-22:
This might be a time zone problem. I noticed that the start epoch time for shot-id 5011 is read as 1435550400 from the event table. 

When I used an online tool to convert this time I found the following:

```
timestamp to human date: 1435550400
GMT: Mon, 29 Jun 2015 04:00:00 GMT
Your time zone: 6/28/2015, 9:00:00 PM GMT-7:00 DST
```

The date in the local web service timezone closely matches to the date being returned by the PH5 Event Web Service. Perhaps the script is using the local timezone, which in the case of where the web services are currently served would be Seattle (PST).

derick-hess, 2017-05-22:
Yeah I will check on this.  I havent checked the event metadata script but the time in the sac file is correct.  I'll update the script and make sure it is returning gmt.

nick-iris, 2017-05-22:
The problem is ph5toexml.py line 219:

`fdsn_time=datetime.utcfromtimestamp(epoch+microseconds).strftime("%Y-%m-%dT%H:%M:%S.%fZ")`

The adding the epoch time to the microseconds in the date conversion is resulting in the wrong date.

nick-iris, 2017-05-22:
Easy fix. Just convert from microseconds to seconds before adding.

`fdsn_time=datetime.utcfromtimestamp(epoch+(microseconds/1000000.)).strftime("%Y-%m-%dT%H:%M:%S.%fZ")`

commit 26130d70 (2017-05-22): Merge pull request #48 from PIC-IRIS/fix_47

Fix #47
commit 69be0848 (2017-05-22): Fix #47
closed 2017-05-22 by derick-hess