PIC-IRIS/PH5 #49: ph5toexml.py needs to enforce deploy/pickup times

nick-iris, 2017-05-22:
Currently ph5toexml.py does not enforce deploy/pickup times but ph5tomsAPI.py does. 

We need to add check event starttimes agains the deploy pickup times, so that user doesn't see events outside of the accepted date range.

@derick-hess and @rsdeazevedo I'm not sure of the best way to read the deploy/pickup times to compare against without reading the entire Array_t table. Ideas?

derick-hess, 2017-05-22:
The array table is usually one of the smallest tables. Array_t contains a pickup and deploy time for the entire array table. SO it would actually only be reading two entries per array to get the deploy and pickup times. I'll  fix this sometime this afternoon

nick-iris, 2017-05-22:
If there are potentially different deploy/pickup dates by array, then will we enforce that the shot must have occured between the min deploy date and max pickup date?

The event web service doesn't currently accept `array` as an argument.

nick-iris, 2017-05-22:
@derick-hess Are you referring only to ph5toexml.py?

derick-hess, 2017-05-22:
Yes. I will test ph5tostationxml on the ph5 server this bug is showing up on as well.

I see: the issue it is reading basepaths as a string instead of a list and the  for basepath in basepaths: line is evaluating it character by character

nick-iris, 2017-05-22:
Good catch. It looks like in ph5tostationxml.py we split on ',' in `__main__`.
```
    if args_dict.get('ph5path'):
        args_dict['ph5path'] = args_dict.get('ph5path').split(',')
```
We need to add that same check to ph5toxml.py.

derick-hess, 2017-05-22:
I'll add that fix to ph5toexml while i'm also fixing the deploy pickup times

nick-iris, 2017-05-22:
Thanks. Good catch. For the web services I'm already passing a list, so it will need to handle that case.

derick-hess, 2017-05-22:
No problem. I'll just take all the arguments and convert them to lists in __main__ so it doesn't interfere with anything web service related

derick-hess, 2017-05-22:
I'll have this all fixed and a commit for review in the morning (maybe tonight). Have the array reading done and it only added a few milliseconds total to each network, just have to finish parsing all the command line arguments to their proper formats.

closed 2017-05-23 by nick-iris