PIC-IRIS/PH5 #54: PH5 web form needs to be updated to access PN4 experiments

nick-iris, 2017-06-02:
The PH5 web form needs to be updated to access the /pn4 directory so that updated (PN4) experiments can remain available in SEG-Y.

A notice should also be added to the form informing users to use the PH5 Web Services for all non SEG-Y data requests.

**Tasks:**
- [ ] Update web form to access PN4 experiments
- [ ] Add a notice to the web form telling users to use the PH5 Web Services

nick-iris, 2017-07-26:
@derick-hess Until everything is updated to PN4, it would be nice to also have an indicator on the[ experiment list page](http://ds.iris.edu/pic-ph5/experiment_list.php?EXP) that shows whether or not a particular experiment is available via web services.

nick-iris, 2017-07-26:
Also, once everything is updated to PN4 I think the experiment list and request form pages should get as much information as possible by making queries to the web services. 

This way we are sure to have the same information in both places.

nick-iris, 2017-07-31:
Derick sorry to hassle you but it would be good to get this updated soon. Are you planning on working on this soon?

derick-hess, 2017-07-31:
Yeah, this will be complete this week

nick-iris, 2017-07-31:
Thanks Derick. The DMC operations group has asked me about this several times.

Once everything is updated to PN4 you may consider creating the table of experiments by making a request to the PH5 web services. This way you don't manually have to maintain the list (assuming that's what you current do).

nick-iris, 2017-08-10:
It would also be nice to include a column of FDSN network codes on the main [experiments list](http://ds.iris.edu/pic-ph5/experiment_list.php).

nick-iris, 2017-08-10:
Never mind there is a column there already it is just mostly empty.

derick-hess, 2017-08-10:
Yeah I think Rick updated a spreadsheet with most of them filld out now. There are only a few missing. I'll update them as I update the webform

nick-iris, 2017-08-10:
Sounds good. The maps feature also no longer works.

derick-hess, 2017-08-10:
I'm rewriting the maps tab to go back to the old way of doing things just using google maps in a <div> since no browser supports the google earth NPI plugin anymore

nick-iris, 2018-02-22:
Any progress on getting the web form updated? It would be best not to have multiple versions of the same data on the server (pn3 and pn4).

In regard to the web form maps, I would strongly suggest just adding hyperlinks to the new GMAP for experiments also available through the web services. For example, the map for the Wavefields experiment can be accessed in the following way:

http://ds.iris.edu/gmap/#network=YW&starttime=2016-06-21&endtime=2016-07-27&planet=earth&datacenter=IRISPH5

derick-hess, 2018-02-22:
The I'll make that a priority to get it done over the next few weeks. GMAP would be great to use, but the only issue with using GMAP is users need to be able to see the events and stations mapped together.

nick-iris, 2018-02-22:
Thanks Derick. That's too bad that GMAP can't be leveraged here. You may still want to consider providing links to it, along side a google map that is native to the web form.

At some point in the future I may add the ability for users to provide links to external data sources.   At such a point, you would be able to use GMAP as a stand alone solution.

derick-hess, 2018-02-22:
Yeah I think on the maps page I'll still link to the GMAP

closed 2019-01-18 by derick-hess