PIC-IRIS/PH5 #74: Station text sensitivity incorrect

derick-hess, 2017-06-29:
when I check the instrument sensitivity  from obspy response object on station 518 ch HHZ in is reported as:

Instrument Sensitivity:
	Value: 944485763.626
	Frequency: 1.0
	Input units: M/S
	Input units description: None
	Output units: COUNTS
	Output units description: None

but the station text reports it as: 

YW|518|01|HHZ|36.5967756197|-97.6347917006|328.42|0.0|0.0|-90.0|Guralp cmg-3t|**_629129.0|0.05|M/S_**|100.0|2016-06-19T00:00:00|2016-10-30T00:00:00

nick-iris, 2017-06-29:
Can you please elaborate on what exactly you are comparing?

I ran the following query and concur that it returns a different value:
http://service.iris.edu/ph5ws/station/1/query?network=YW&level=channel&format=text&station=518&nodata=404

I'm guessing that there is either a difference in the response file in PH5, or a bug in ph5tomsAPI.py where it is assigning the wrong value to scale and scale frequency.

nick-iris, 2017-06-29:
The code in ph5tostationxml.py computes the combined response of the datalogger and sensor if both are present. Is this what you did when you read the response file?

```
            response_file_das_a_name = Response_t.get('response_file_das_a', None)
            response_file_sensor_a_name = Response_t.get('response_file_sensor_a', None)
            # parse datalogger response
            if response_file_das_a_name: 
                response_file_das_a = self.ph5.ph5_g_responses.get_response(response_file_das_a_name)
                dl = Parser(response_file_das_a)
            # parse sensor response if present
            if response_file_sensor_a_name:
                response_file_sensor_a = self.ph5.ph5_g_responses.get_response(response_file_sensor_a_name)
                sensor = Parser(response_file_sensor_a)
    
            inv_resp = None
            if response_file_das_a_name and response_file_sensor_a_name:
                # both datalogger and sensor response
                comp_resp = Parser.combine_sensor_dl_resps(sensor=sensor, datalogger=dl)
                inv_resp = comp_resp.get_response()
            elif response_file_das_a_name:
                # only datalogger response
                inv_resp = dl.get_response()
            elif response_file_sensor_a_name:
                # only sensor response
                inv_resp = sensor.get_response()
```

derick-hess, 2017-06-29:
Yeah I think teh station txt code may be grabbing the wrong variable when it prints (havent looked at it)

when I print inv_resp.instrument_sensitivity

the reported instrument sensitivity is correct but thats not what is being written in the station txt output.

nick-iris, 2017-06-29:
@derick-hess Could you email me the response files that are currently in the Wavefields PH5 Experiment?

derick-hess, 2017-06-29:
sure I'll send them to you .  Steve did say that LLoyd noted that way Antelope calculates sensitivity (what fdsn web services use) vs his code is a bit different so when combining responses into one the value and frequency may be be different but they are equivalent. 

His software is being made to replace antelope in the future.

nick-iris, 2017-06-29:
For me, printing inv_resp.instrument_sensitivity prints:

```
sensitivity = Instrument Sensitivity:
	Value: 629129.0
	Frequency: 0.05
	Input units: M/S
	Input units description: None
	Output units: COUNTS
	Output units description: None
```

This leads me to believe that it is not a problem with the StationTXT print function.

nick-iris, 2017-06-29:
The individual datalogger and sensor responses read are:

```
datalogger resp = Channel Response
	From M/S () to COUNTS ()
	Overall Sensitivity: 629129 defined at 0.050 Hz
	11 stages:
		Stage 1: PolesZerosResponseStage from M/S to V, gain: 1
		Stage 2: ResponseStage from V to V, gain: 1
		Stage 3: CoefficientsTypeResponseStage from V to COUNTS, gain: 629129
		Stage 4: CoefficientsTypeResponseStage from COUNTS to COUNTS, gain: 1
		Stage 5: CoefficientsTypeResponseStage from COUNTS to COUNTS, gain: 1
		Stage 6: CoefficientsTypeResponseStage from COUNTS to COUNTS, gain: 1
		Stage 7: CoefficientsTypeResponseStage from COUNTS to COUNTS, gain: 1
		Stage 8: CoefficientsTypeResponseStage from COUNTS to COUNTS, gain: 1
		Stage 9: CoefficientsTypeResponseStage from COUNTS to COUNTS, gain: 1
		Stage 10: CoefficientsTypeResponseStage from COUNTS to COUNTS, gain: 1
		Stage 11: CoefficientsTypeResponseStage from COUNTS to COUNTS, gain: 1
 WARNING (norm_resp): computed and reported sensitivities differ by more than 5 percent. 
	 Execution continuing.
sensor resp = Channel Response
	From M/S () to COUNTS ()
	Overall Sensitivity: 1500 defined at 1.000 Hz
	2 stages:
		Stage 1: PolesZerosResponseStage from M/S to V, gain: 1500
		Stage 2: CoefficientsTypeResponseStage from V to COUNTS, gain: 1
 WARNING (norm_resp): computed and reported sensitivities differ by more than 5 percent. 
	 Execution continuing.
```

nick-iris, 2017-06-29:
Can you explain how you arrived with this output?

```
Instrument Sensitivity:
Value: 944485763.626
Frequency: 1.0
Input units: M/S
Input units description: None
Output units: COUNTS
Output units description: None
```

derick-hess, 2017-06-29:
Yeah looks like somehow the station txt is only reading the datalogger sensitivity and not the combined sensitivity

I'm printing the sensitivity at this point:

```
if response_file_das_a_name and response_file_sensor_a_name:
                # both datalogger and sensor response
                comp_resp = Parser.combine_sensor_dl_resps(sensor=sensor, datalogger=dl)
                inv_resp = comp_resp.get_response()
                print inv_resp.instrument_sensitivity
```

nick-iris, 2017-06-29:
Yes but printing there does not output:

```
Instrument Sensitivity:
Value: 944485763.626
Frequency: 1.0
Input units: M/S
Input units description: None
Output units: COUNTS
Output units description: None
```

it outputs

```
Instrument Sensitivity:
Value: 629129.0
Frequency: 0.05
Input units: M/S
Input units description: None
Output units: COUNTS
Output units description: None
```

How did you get the seemingly "correct" value of 944485763.626?

derick-hess, 2017-06-29:
Oh, I just realized I grabbed the latest RESP code form obspy last night becasue the server I'm workign on didnt have it installed and Lloyd made a few commits about a week ago. 

s://github.com/obspy/obspy/pull/1185/

Maybe that is the difference?

nick-iris, 2017-06-29:
That fixed it! I just pulled the latest version and sure enough the values match. Good catch!

http://service.iris.edu/ph5ws/station/1/query?network=YW&level=channel&format=text&station=518&nodata=404

derick-hess, 2017-06-29:
Yeah, Looks like before it was just grabbing the datalogger sensitivity instead of combined sensitivity. The calculated values/frequency will be slightly different than the FDSN but they should be equivalent.

nick-iris, 2017-06-29:
I think we should do some more thorough tests before closing this. The recorded instrument sensitivity in MDA is: **Sensitivity	9.466290e+08 @ 3.000e-01 Hz (SEED Stage 0)**
http://ds.iris.edu/mda/YW/518/01/HHZ?timewindow=2016/06/19%2000:00:00-2016/12/30%2023:59:59

The value we get is at a different frequency: 9.444858e+08 @ 1.0 Hz

nick-iris, 2017-06-29:
We need to make sure these are equivalent for this response:
9.466290e+08 @ 3.000e-01 Hz
same as
9.444858e+08 @ 1.0 Hz

derick-hess, 2017-06-29:
They should be but I'll ask lloyd to check. 
He addressed the difference in sensitivity calculations in a meeting about his software last week but said given the same Responses are loaded the sensitivities will be equivalent. Antelope outputs the sensitivity in odd frequencies when it calculates all the combined values.

nick-iris, 2017-06-29:
@chad-iris I'm seeing ~0.3% difference between the SEED and PH5 Wavefields response amplitude at 1Hz.

From evalresp we get an amplitude of 9.474110E+08 @ 1.07Hz

http://service.iris.edu/irisws/evalresp/1/query?output=fap&sta=518&net=YW&loc=01&cha=HHZ&time=2016-171T00:01:00

while from the PH5 Station Web Services we get an amplitude of 9.444858e+08 @ 1.0 Hz

% difference = (9.47 - 9.44) / 9.44 = 0.0031 ≈ 0.3%

Is this significant?

derick-hess, 2017-06-29:
I asked the data group. This comes from the responses in antelope being tweaked a bit for passcal specific instruments as compared to the the NRL. Data group said it shouldnt make a difference.

Though i did notice the the Infrasound stations (in the 600s) have the correct value/frequency but  incorrect units. Units should read PA not M/S. Pretty sure this is from LLoyds obspy code and may just be an oversight since infrasound is rarely used.

derick-hess, 2017-06-29:
The responses look to be matching with very slight differences in values. THis is for station 518 HHZ

Here is the FDSN vs PH5 response plotted over each other

https://www.passcal.nmt.edu/~dhess/518_HHZ_response.png

and here is the csv

https://www.passcal.nmt.edu/~dhess/FDSNvsPH5.csv

nick-iris, 2017-06-29:
Thanks. Can you open up permissions on https://www.passcal.nmt.edu/~dhess/FDSNvsPH5.csv?

 I get a 403 Forbidden error.

derick-hess, 2017-06-29:
changed the link. Our web server wouldnt allow csv files. Uploaded to our owncloud server and updated the link

derick-hess, 2017-06-29:
one thing to note: a few of the lower frequency values are a higher % off because I couldn't find as close matching frequencies for the lower values.

nick-iris, 2017-07-07:
@rsdeazevedo  and @derick-hess For Wavefields I used the evalresp web service to compare the PH5 instrument sensitivity to what is in the miniSEED archive for each channel type. I then calculated the % difference for each at the given frequency.

| Channel Type | PH5 Sensitivity | MSEED Archive Sensitivity | % Difference |
| ------------- | ------------- | ------------- | ------------- |
| LH | 9729.19760913 @ 1Hz | 9768.776E @ 1Hz | 0.41% |
| HH | 944485763.626 @ 1Hz | 947331500 @ 1Hz | 0.30% |
| DP | 1029629991.04 @ 40Hz | 1029630000 @ 40Hz | 0% |
| HDF | 12585.5577659 @ 0.5Hz | 12777.39 @ 0.5Hz | 1.50% |

derick-hess, 2017-07-08:
one thing that is odd is that the DP channels not being exactly the same from evalresp. They are the exact same RESP file being used with no calculations needing to be done since there is no combining of multiple resp files.

Bruce said the difference isn't significant and will leave it up to the DMC and BRTT(the company that makes Antelope, the software currently used for response files) to figure out who's response information is correct since PH5 is using the NRL for response information.

derick-hess, 2017-07-08:
I think the next step is we need to be able to look at the rt130 response info and the cmg-3t response info that is being used for the miniseed before it is combined and compare it to the response info from teh NRL. 

If they are different then we know the difference is most likely in the actual response information, if they are the same then we know it's a calculation/combining issue and we will want to copy all the relevant info from this issue over to an issue on obspy so it can be addressed there and we can help get it fixed.

derick-hess, 2017-07-27:
Looking at the testing Lion just did with the obspy RESP I think it's safe to say that all the calculations are correct. He ran it against the entire IRIS and ORFEUS inventories grabbing the response from web service, reading it into an inventory, writing out to stationxml, read it back into an inventory, then checked that the response from evalresp, SEED, stationxml and from RESP all matched.

So it seems the difference is in the actual responses used in antelope vs what is in the NRL.

closed 2017-07-31 by nick-iris