PIC-IRIS/PH5 #75: StationXML total number of channels is incorrect

nick-iris, 2017-07-06:
The total number of channels in the PH5 StationXML is incorrect.

i.e. the total number of channels for YW station 1001 shows 397 
https://service.iris.edu/ph5ws/station/1/query?format=xml&level=station&network=YW&starttime=2016-06-21T16:43:57&endtime=2016-06-26T16:43:57&nodata=404
when it should be 3.

nick-iris, 2017-07-06:
Looks like ph5tostationxml.py line 504 is the culprit.

https://github.com/PIC-IRIS/PH5/blob/refactor/ph5/clients/ph5tostationxml.py#L504

derick-hess, 2017-07-06:
Yeah, Not sure how that happened. I'll fix it.

nick-iris, 2017-07-06:
Changing
`obs_station.total_number_of_channels = len(sta_list)`
to 
`obs_station.total_number_of_channels = len(station_list)`
should do the trick.

nick-iris, 2017-07-06:
We might want to change `station_list` to `deployment_list` since it is a bit confusing...

closed 2017-07-07 by nick-iris