PIC-IRIS/PH5 #81: PH5toStationXML Response Management

derick-hess, 2017-07-11:
We need to add a key for sample rate. Since there are different responses based on sample rate that can have the same sensor_key and data_logger key, another ket based on sample_rate needs to be added.

I found this when the VM stations were added. There is now response for VM currently in the PH5, but I noticed since they are still cmg-3t/rt130 instruments, it's using the last loaded response and loading that into the inventory object as the response for the VM channels.

I think this error would only ever come up if there is no response for a channel and it matched keys from a previous channel.

nick-iris, 2017-07-11:
This shouldn't be a problem to add. You can just add them to the sensor_keys. The ResponseManager just checks if a response was already read, so that the program isn't reading the RESP file for every channel.

I think something like this on ph5tostationxml.py lines 364-366 would be all that is needed:
```
        sensor_keys = [obs_channel.sensor.manufacturer,
                       obs_channel.sensor.model,
                       obs_channel.sample_rate]
```

Note the addition of the sample rate the sensor_keys list.

derick-hess, 2017-07-11:
Thanks. I'll fix it and run some tests today.

commit cbd2b37e (2017-07-13): Merge pull request #82 from PIC-IRIS/fix#81

fix #81
closed 2017-07-14 by derick-hess