PIC-IRIS/PH5 #83: Missing LICENSE.txt file

nick-iris, 2017-07-13:
The ph5 package is missing a LICENSE.txt file. Is there a particular license that you want to use?

There are some helpful guidelines here: https://opensource.guide/starting-a-project/#choosing-a-license

We'll want to add this before publishing to [PyPI](https://pypi.python.org/pypi).

derick-hess, 2017-07-13:
We sue the MIT license. I'll add it back in.

nick-iris, 2017-07-13:
Okay it should be added to the _setuptools_build_ branch in the /ph5 directory. A symlink to this file should also be added on the top level page.

You can use the /ph5/CONTRIBUTORS.txt as a reference.

nick-iris, 2017-07-14:
Thanks Derick.  I added line breaks to the license so it is easier to read and also a symlink to the top level of the project.

closed 2017-07-14 by nick-iris