PIC-IRIS/PH5 #86: Package and Module names should be lower case

nick-iris, 2017-07-19:
A lot of the modules in the PH5 package contain upper case letters. These should be changed to all lower case.

The PEP8 Style Guideline states the following regarding module and package names:
> Modules should have short, all-lowercase names. Underscores can be used in the module name if it improves readability. Python packages should also have short, all-lowercase names, although the use of underscores is discouraged.
> 
> When an extension module written in C or C++ has an accompanying Python module that provides a higher level (e.g. more object oriented) interface, the C/C++ module has a leading underscore (e.g. _socket).

nick-iris, 2017-07-19:
Additionally, the "PH5View" package should be changed to "ph5view".

derick-hess, 2017-07-19:
working on it now.

nick-iris, 2017-07-19:
Cool. In addition to updating all related imports to reflect the new file names, the entry_points in setup.py will have to be updated as well.

closed 2017-07-19 by nick-iris