PIC-IRIS/PH5 #88: Duplicate code should be moved to ph5utils.py

nick-iris, 2017-07-19:
There is a lot of duplicate code between the utilities. This code should be pulled into the ph5utils.py core module.

Some of the heavily duplicated classes I found were: _Index_t_Info_, _Resp_, and _Rows_Keys_.

nick-iris, 2017-07-19:
@rsdeazevedo and @derick-hess What are your thoughts about moving the commonly used Index_t_Info, Resp, Das_Groups, Offset_Azimuth, and Rows_Keys to ph5.core.ph5utils?

derick-hess, 2017-07-19:
That's probably a good idea. I haven't take a look at those methods yet but it would probably be wise to put them in ph5utils or ph5api (wherever they fit best).

It might be a while before I have time to work on this. I think I need to dedicate the next week or so to the webform and the RESP loading utility so those are good to go as well.

nick-iris, 2017-07-19:
Sounds good. I agree that the web form should be a relatively high priority. I might make the change and create a new pull request for you to review since it won't take too long. Thanks.

ascire-pic, 2020-10-23:
Closing this issue due to beginning development of GeoHDF

closed 2020-10-23 by ascire-pic