PIC-IRIS/PH5 #94: Add restricted status to ph5tostationxml.py

nick-iris, 2017-07-31:
The StationXML produced my ph5tostationxml.py should include the restricted status.

For non-restricted experiments the status would be `restrictedStatus="open"`. For restricted experiments that status would be `restrictedStatus="closed"`.

derick-hess, 2017-07-31:
Is there an easy way to get the restricted status from a web service? We could add code into ph5tostationxml to grab that information then write it into a custom attribute.

nick-iris, 2017-07-31:
Yeah I think it would be pretty easy since I'm already passing a list of restricted SNCL starttime/endtime segments to ph5tostationxml.py. I'll work on adding this.

nick-iris, 2017-07-31:
Actually, I was mistaken. We don't pass the restricted list to the station service. It gets passed from the Dataselect Web Service to ph5tomsAPI.py. I'll work on adding a the ability to also pass this information to ph5tostationxml.py

nick-iris, 2017-07-31:
I ended up making this fix entirely in the Web Services code. Adding this feature to the core API would require it to access our internal Restricted Web Service.

derick-hess, 2017-07-31:
Okay makes sense. Thanks Nick.

nick-iris, 2017-08-01:
I thought I had this worked out, but the challenge is that our internal restrict web service only accepts request at the channel (SNCL-ST/ET) level, and _not_ at the station or network level.

Currently the fix I've written requires always making a channel level request. This isn't very efficient so I'm trying to think of a work around.

I'm going to bring this up at our DMC Web Services meeting tomorrow and see if we can collectively 
come up with a solution. Reopening since this is unfortunately still in progress.

derick-hess, 2017-08-01:
Okay thanks. The last resort would be to add a restricted flag to the experiment table. This wouldnt be the best option since it would require someone to manually change those entries as restricted status changed or some kind of cron script to run to check and update.

nick-iris, 2019-06-07:
This will be addressed in the new station service that is currently under development.

ascire-pic, 2020-10-23:
Issue is being addressed in station web service development rather than within the PH5 framework. Closing this issue as the functionality is being implemented outside of PH5.

closed 2020-10-23 by ascire-pic