PIC-IRIS/PH5 #96: Add common shot gather example to the PH5 Wiki

nick-iris, 2017-07-31:
We should add a example to the [PH5 Web Services section](https://github.com/PIC-IRIS/PH5/wiki/PH5-Web-Services#requesting-data-with-obspy) of the PH5 Wiki that explains how to create a common shot gather using ObsPy with the PH5 web services.

rsdeazevedo, 2017-07-31:
Agreed. If anything, i think this may stimulate positive discussions 
from the communities.

nick-iris, 2017-08-08:
I figured out that we actually can use the ObsPy get_events method to retrieve event metadata from the ph5ws-event web service. I added an example to the PH5 Wiki:

https://github.com/PIC-IRIS/PH5/wiki/PH5-Web-Services#ph5-event-web-service-request-with-obspy

 I'm currently working on writing an example of how all three service could be for a common shot order gather. If you have any ideas of how this might work, I'd appreciate your input. I'll post my example once finished.

nick-iris, 2017-08-09:
Is the convention for a common shot order gather to collect a certain amount of data for each channel before & after each shot start-time (like we do for the web service "by shot" request type) and then order that data by the station-id? Or do you order the data by the physical distance that the sensor was from the shot? Or something different entirely?

I'm trying to figure out given a catalog of event (shot) metadata and an inventory of station metadata, how to request the waveform data for a common shot order gather. I'll then use this as an example for others.

derick-hess, 2017-08-09:
So given a shot position and time you gather data for each  station starting at the shot time. The data is ordered based on physical distance from the shot.  IN the image below each vertical line is a station with time starting from the top of the image. You can see there are stations to the left and right of the stations showing their relative distance from the shot

[image]

I haven't done it in obspy in a while, but you can just request the data like normal into an inventory then attach the shot time and shot location and obspy can do all those calculations and output a gather for you.  There are also other standard things you can do like applying a reduction velocity. 

I haven't looked at it but Lans PH5view code creates shot gathers from PH5. I think it is all hand coded by her and doesnt use obspy though

nick-iris, 2017-08-09:
Unfortunately it looks like the ability to add event and station metadata to a trace object hasn't been implemented yet. See the following thread: https://github.com/obspy/obspy/issues/618

Perhaps you know of another way to do this?

derick-hess, 2017-08-09:
It looks like before what I was doing was specifically in a "section" plot using stream.plot
https://docs.obspy.org/packages/autogen/obspy.core.stream.Stream.plot.html

That would work if you collected all the traces into a stream cut on the start time of the  event.  Attached coordinates to each trace and then passed the event coordinated into the plot using ev_coord tuple

SO looks like you can attach coords to traces but not actually attach the event coords to the stream, only plot based on it.

nick-iris, 2017-08-09:
Got it working! Here is a shot gather for network ZI.

[image]

derick-hess, 2017-08-09:
awesome!

nick-iris, 2017-08-09:
I added an new page to the wiki here: https://github.com/PIC-IRIS/PH5/wiki/PH5-Web-Services-Shot-Gather

closed 2017-08-16 by nick-iris