PIC-IRIS/PH5 #97: Version tagging

derick-hess, 2017-07-31:
I have looked over the PH5 document we sent to Kent and the working group. It was decided that the major version number of the software would correspond with the major version number of the PH5 format. IN the original document version  Software version 4.0 would be the initial release with 4.1 being released when RESP was integrated into PH5. The thought on this was it would take much longer for the changes to be in Obspy.

I think we should release 4.0 when the current 2 pull requests(#92 and #93) are done. Incremental number releases for bug fixes such as 4.0.1 and new features not related to the next major release of 5.0 following the 4.1.xxx format

PH5 version 5.0 and software has the following possible additions: 
* Adding provenance model(seis-provenance)
* Additions to event table to store earthquake metadata

nick-iris, 2017-07-31:
Thanks for looking into this. Currently the version is set to 4.0.0 in setup.py. See https://github.com/PIC-IRIS/PH5/blob/master/setup.py#L26.

This is where the overall PH5 version should be tracked as it is what will get displayed as the installed version. For the initial release no change should be necessary besides creating and adding a note to a CHANELOG.txt file.

rsdeazevedo, 2017-07-31:
Also, i believe that the document we sent to Kent included full implementation of the Guides_g in version 5.or toward the end of 4 releases. This will define ph5 in terms of legacy data formats.

closed 2017-08-04 by derick-hess