PIC-IRIS/PH5 #113: Output metadata in csv format

derick-hess, 2017-08-24:
David Okaya suggested having a new tool or ph5tostationxml output the metadata in basically a csv file where the user can set the delimiter. He said this would be really useful for use in many active source software tools as well as really useful for creating maps using Generic Mapping Tools(GMT).

This would be an easy addition to ph5tostationxml.

nick-iris, 2017-08-24:
Tim Ahern mentioned that he would like to have a discussion about where to go next with the PH5 Web Services. He'll be back in two weeks so I think we should wait till then before implementing any additional outputs.

derick-hess, 2017-08-24:
In the meeting this afternoon, I showed him the stationtext and he said that yes, something like that is what he would like, with the main differences really only being array # is needed for active source people and the ability to select the delimiter you want. He said most of the software is smart enough to select the columns you want to use but some require specific delimiters.  

It would also be useful to have it available in the a field use version of the software.

rsdeazevedo, 2017-08-24:
I agree with Nick. Things like this that will affect other users need to be vetted carefully by all parties involved, including Tim and Bruce.

derick-hess, 2017-08-24:
Yeah I think mainly David wants to be able to output something like station text in the field and currently you cant do that

derick-hess, 2017-08-24:
It also may suffice to just output stationtxt on the command line for field use.

derick-hess, 2017-11-07:
This was actually addressed incidentally in a previous pr. TEXT format is now available via the command line of ph5tostationxml. Closing

closed 2017-11-07 by derick-hess