PIC-IRIS/PH5 #148: Tool needed to update node das_t

derick-hess, 2017-11-01:
A command line tool is needed that will allow a user to assign a new serial number to a node id that was accidentally deployed multiple times in the field.

The command line tool would allow the user to specify the raw file name of the node that needs to be renamed and supply a new serial number. 

Example: 
Node 104 array 1 was deployed twice creating file names:
R1_104.0.0.rg16
R1_104.0.1.rg16

once in ph5 they would both be listed as serial number 1X104 but only data for the first  file would ever be extracted so R1_104.0.1.rg16 would need to be renamed to a new unique node ID

example command could be:

`das_rename -f R1_104.0.1.rg16 --new_name 1004`

This would go through the das tables and wherever the original raw file was R1_104.0.1.rg16 it would update the serial number to now be 1X1004

The user would still need to create a new entry in the array table with the correct metadata for station 1004. This would be determined most likely only by hand written deploy notes from the field.

CeliaPacheco, 2018-06-21:
In what group does the serial number need to be updated? Right now what I have is I take in the raw file name and search for the das serial number associated to that file name. But now that I have that serial number I'm not sure in what group it needs to be changed. Looking at the documentation I can see that the das serial number is in both the Receivers Group and Sorts Group. Do I change the serial number in both?

derick-hess, 2018-06-21:
It should be in both. I'm going to have to think about it again because I think in the Receivers group a new das_g_XXX where XXX is serial will need to be created. I'll think about this and draw out a diagram on what should happen. I think a diagram will make it make sense.

timronan, 2020-12-22:
This ticket may be pointing towards a bug where data from a node that is deployed multiple times with different configurations, different epochs, cannot be be accessed because one of the node's configurations is being covered up by metadata information from the other configuration. Look in the DAS table for slight variants in a nodes serial number. A das table with multiple nodal serial numbers that are slightly different may provide evidence, or a test case, that can either confirm or deny this ticket.