PIC-IRIS/PH5 #150: RESP upload utility needs to be added

nick-iris, 2017-11-03:
There is a great need for a RESP upload utility. Currently we have been receiving experiments from PIs without RESP information, and @derick-hess has been adding this info for them. This works as a temporary solution, but makes it so that outside users cannot use all of the extraction programs before sending us their data. For example ph5tostationxml.py will not work without the RESP info.

derick-hess, 2017-11-03:
Resp_load has already been developed and added PR #136

derick-hess, 2017-11-03:
I do need to add some instructions. Also, I will contact our data group and see if the long doc and new ph5 in a nutshell is reayd so I can link to them on the github

nick-iris, 2017-11-03:
Thats great thanks Derick!

closed 2017-11-03 by nick-iris