PIC-IRIS/PH5 #152: Make PH5 nuke_table automatically create backups

nick-iris, 2017-11-06:
I think it would be a nice feature for the PH5 nuke_table command to automatically create a KEF backup of the table before deleting it's contents. For advanced users this feature could be optionally ignored.

For example:

`nuke_table -n master.ph5 -p /hdf5-data2/PH5_Experiments/pn4-stage/17-028/ -A 001`

Could create a file Array_t_001.kef.bk before deleting its contents.

With the current implementation, it seems really easy to accidentally delete something without having a backup of the original table contents.

rsdeazevedo, 2017-11-06:
Nick, that is a good idea. I'll do that later this week. I am 
implementing external shot files as requested by David which should be 
done today.

nick-iris, 2017-11-06:
Thanks Steve.

I had a few more ideas about this:

- [ ] It would be good to warn the user if a backup file already exists with the same name before creating a new one. You don't want the program to overwrite another kef file in their experiment directory without asking the user first.
- [ ] Backup files should be timestamped, to further prevent overwriting existing backup files.
- [ ] Backup files should include the experiment report number, so that the can be recognized outside of the context of the given experiment.
- [ ]  Backup files should probably be written to a subdirectory that also contains the report number. (e.g. such as a directory named backup-16-015/)
- [ ] If a user lacks the permissions for the nuke_table command to create a backup file, they should at the very least be warned, and have an option to cancel the destructive operation. The program should handle this edge case gracefully.

rsdeazevedo, 2017-11-15:
Backup nuke should resolve all ideas except those requiring report number.

closed 2017-11-15 by rsdeazevedo

nick-iris, 2017-11-15:
Thanks Steve, for this very useful addition.