PIC-IRIS/PH5 #172: ShotText format missing type field

nick-iris, 2017-12-18:
The shottext output format in ph5toexml does not include the "type" field that is found in the QuakeML output. What do you think of adding this field to the shottext output format?

derick-hess, 2017-12-18:
Seems like a good idea to me

nick-iris, 2017-12-19:
Looking into this further, I've found that the "type" field is being hardcoded to "controlled explosion". 

See https://github.com/PIC-IRIS/PH5/blob/master/ph5/clients/ph5toexml.py#L590. 

In my opinion, this is misleading since the shots are not necessarily from explosions (as we found with experiment 12-003). 

Perhaps we need to add a shot "type" field to the `Sorts_g/Event_t_###` table in PH5? This way users can specify the type of shots that are being used. Alternatively we could use the `/Experiment_g/Sorts_g/Event_t_###/description_s` field, but this is already being defined separately in the QuakeML.

derick-hess, 2017-12-19:
We would have to go through the PH5 working group to add any new fields I think. I do think it needs to be added to the event table though

derick-hess, 2017-12-19:
QuakeML 2.0 also has more features that will be useful to us. It adds a lot more for doing non earthquake events

ascire-pic, 2020-10-23:
Related to issue #210 

Event type is not a field that is currently recorded in PH5. Would need to add type to the event metadata/shot table