PIC-IRIS/PH5 #177: ph5toms file output naming

derick-hess, 2018-02-01:
Feature addition to allow control of file naming. Possibilities include either a choice of a few different naming schemes or a more robust flag that allows you to choose the output name formatting.

nick-iris, 2018-03-24:
By default I think all data files should be in the format:

**YW.1001..DP1.2016-06-21T16:43:57.832000.mseed**

I need to use this format for the web services in order to match the other services, and it would be nice to remain consistent everywhere. I'm going to be pushing a change soon that will have this included. I'm adding a GeoCSV data output format for the web services that required some changes to ph5toms, including this.

CeliaPacheco, 2018-05-21:
What type of control is needed for file naming? I'm looking over the code now and I see 5 methods that are for 5 different types of naming schemes but only mseed and sac are implemented. Should I implement the other 3?

derick-hess, 2018-05-21:
Just the mseed and sac file name generation.

the format nick suggested is and looking is already implemented:
[net_code].[seed_station].[seed_loc].[seed_ch].[start_time]

That shouls be the standard and  a few users have asked for us to add an argument to arg_parse that is:
[array].[seed_sta].[seed_channel].[start_time]

derick-hess, 2018-06-07:
adressed in pr #205

closed 2018-08-10 by derick-hess