PIC-IRIS/PH5 #182: kefedit ver. 2018.037 bugs

nick-iris, 2018-02-20:
Using kefedit for the first time I've found the following bugs:

- Upon trying to open a master.ph5 file from the tool, I get the following error:
```
Traceback (most recent call last):
  File "/Users/nick/Development/workspace/PH5/ph5/utilities/kefedit.py", line 247, in OnOpenPH5
    self.ph5api, availTables, arrays, shotLines, offsets, das = kefutility.GetPrePH5Info( self.filename, self.path2file)
ValueError: need more than 5 values to unpack
```
- Closing the program I get the following error:
```
Traceback (most recent call last):
  File "/Users/nick/Development/workspace/PH5/ph5/utilities/kefedit.py", line 180, in closeEvent
    sys.exit(application.exec_())
NameError: global name 'application' is not defined
```

derick-hess, 2018-02-21:
@damhuonglan I get the same error. Tested on my mac.
My first guess is that it is crashing when PH5 experiments don't have shotlines or offsets. 

I think the second error happens because it is crashing before the application gets created. That should definitely be handled with a try except and have that error caught.

nick-iris, 2018-02-27:
@derick-hess Were these fixed in PR #183 or is KefView a different program?

derick-hess, 2018-02-27:
I htink Lan just named the branch wrong. PFR #183  addressed the kefedit bug

closed 2018-02-27 by derick-hess