PIC-IRIS/PH5 #207: PH5 to python3

derick-hess, 2018-08-10:
PH5 needs to be converted to python 3 before python 2 end of life. It should be made compatible with both python 2 and 3

El-minadero, 2019-06-19:
This right here is a major barrier for using this format. Almost all my fellow gradstudents operate in a pure python 3.x environ, and are unlikely to adopt any standard that relies on legacy python.

timronan, 2020-12-21:
Although python 2.X is [depreciated](https://www.python.org/doc/sunset-python-2/), it does not seem worth the effort to update PH5 to python 3 unless the update becomes necessary to migrate the data out of PH5. If there is a specific reason to migrate to python 3 this issue can be revisited, but should be considered low priority until we find a justification for this effort.

timronan, 2021-03-25:
My previous comment was short sighted and likely incorrect. As stated by @El-minadero, failing to update PH5 out of a legacy Python version will greatly limit its users, and may end up making PH5 unusable before it's planned end of life. This issue needs be re-evaluated and is not an enhancement. 

The upgrade from python 2 to python 3 is about maintaining control over the software that is actively being used both internally and externally. Failing to perform an upgrade is poses security risk to the stored PH5 data and our PH5 services. This update should be considered soon as python2 has already been depreciated for over a year.

timronan, 2021-03-29:
It seems like Obspy may quit supporting python 2.

https://github.com/obspy/obspy/pull/2577
https://github.com/obspy/obspy/issues/2228 

PH5 is dependent on Obspy and should likely stay in sync with this dependent library.

timronan, 2021-04-06:
Some documentation on how to support python 2 and python 3 simultaneously. https://docs.python.org/3/howto/pyporting.html