PIC-IRIS/PH5 #212: PH5 availability

derick-hess, 2018-08-14:
Program to get the available data down to the sample including gaps.

derick-hess, 2018-08-16:
This program will be used to create an availability service for PH5 similar to https://service.iris.edu/irisws/availability/1/

Inputs:
Master file name
path
array (wild cards allowed)
station id (wild cards allowed)
seed station name (wild cards allowed)
channel id (wild cards allowed)
start time (passcal time format and fdsn time format)
end time (passcal time format and fdsn time format)
output (either stdout or to file (txt, geocsv, json, sync, request)
flag to use deploy/pickup times or to search across all time in experiment
flag for extent or query

functionality:
Extent:
Produces lists of available time extents (earliest to latest) for selected channels (network, station, location and quality) and time ranges.

For every channel check the das table and get the start and end time of the data. The end time is calculated by checking the start time of last recording window and the number of samples recorded and calculating the end time.

Query:
Produces lists of contiguous time spans for selected channels (network, station, location, channel and quality) and time ranges.

For every channel check the das table and loop over entire das table making an entry for every contiguous time span. Gaps trigger a new entry.

nick-iris, 2019-01-28:
There has been some discussion about repurposing the existing [IRISWS Availability web service](https://service.iris.edu/irisws/availability/1/) for  PH5. The irisws-availability service retrieves availability statistics using the [IRIS TSIndex database](https://github.com/iris-edu/mseedindex/wiki/Database-Schema#sqlite-schema-version-11).

We could develop a program that writes availability timespans to a  sqlite tsindex database. We could then use this database to standup a high performance availability service for PH5.

I think it would still be a good idea to write a `ph5availability` command that retrieves availability information directly from PH5. Perhaps this command could act as the source to a hook that writes availability information to a tsindex database. This way local PH5 users can use the PH5 command, while the web services can run off of the tsindex database for higher performance.

derick-hess, 2019-01-28:
Sounds good. I started the availability code today.
Right now I'm writing a a new method to ph5api called get_extent()
It accepts das serial and channel component and returns the earliest and latest times for that channel.

nick-iris, 2019-01-28:
I recently wrote a [tsindex client](https://github.com/obspy/obspy/pull/2206) that is scheduled to be included in ObsPy version 1.2.0. It includes the following methods relating to availability in its API. You may wish to include similar methods in PH5.

Method | Return Type | Description
-- | -- | --
get_nslc(network, station, location, channel, starttime, endtime) | list(tuple) | Return a list of tuples [(net, sta, loc, cha),...] containing information on what streams are included in the tsindex database.
get_availability_extent(network, station, location, channel, starttime, endtime) | list(tuple) | Return a list of tuples [(net, sta, loc, cha, earliest, latest)] containing data extent info for time series included in the tsindex database.
get_availability(network, station, location, channel, starttime, endtime,include_sample_rate=False, merge_overlap=False) | list(tuple) | Return a list of tuples [(net, sta, loc, cha, start, end),...] containing data availability info for time series included in the tsindex database.If include_sample_rate=True, then a tuple containing the sample rate [(net, sta, loc, cha, start, end, sample_rate),...] is returned. If merge_overlap=True, then all time spans that overlap are merged.
get_availability_percentage(network, station, location, channel, starttime, endtime) | 2-tuple of percentage of available data (`0.0` to `1.0`) and number of gaps/overlaps. | Get percentage of available data for a specified network, station, location, channel, starttime, endtime combination.
has_data(network, station, location, channel, starttime, endtime) | boolean | Check if specified stream has any data using the tsindex database.

derick-hess, 2019-01-28:
Thanks. The methods being added to the API are a little more low level but the utility I will write will match those.

derick-hess, 2019-02-08:
@nick-iris I am looking at the tsindex database scheme from the iris-edu github. We should figure out how to handle the differences in PH5 versus what this is meant for. Specifically the two that don't work with PH5 are byteoffset and timeindex.

Since PH5 is broken up into original recording windows, except when something like miniSEED, the need for these is different.  Recording windows are numbered and contain data on the start time of the windows, number of samples, and sample rate.  So after a gap occurs the next Available data will always start at the beginning of a new recording widow. So window number for the start of a new data span might be useful.  Contiguous data can also span multiple windows before a gap (if any) occurs found.
byteoffset also isn't really useful for PH5. PH5 data is stored as an array. Array wouldn't be that useful since of course the data will always start at [0] in each window and array length will be the number of samples in the window with each window starting again at 0.  

byteoffset: could be window number
bytes: could be the same but not all that useful. Number of samples could be useful here.
timeindex: not sure what we want to do here...any ideas?

nick-iris, 2019-02-08:
@derick-hess We don't need to completely fill in the entire tsindex database for the availability web service to work. It only uses the following fields:

- network
- station
- location
- channel
- quality
- updated
- starttime
- endtime
- samplerate
- timespans
- timerates

derick-hess, 2019-02-08:
@nick-iris Thanks. That simplifies things.

commit 44b31c41 (2019-06-04): WIP PH5 data availability (#259)

Add ph5availability client and availability methods to ph5api. Fix #212
closed 2019-06-04 by nick-iris