PIC-IRIS/PH5 #215: Modules should not call sys.exit()

nick-iris, 2018-09-25:
There are lots of places where modules call `sys.exit()`. This is considered a poor practice, as it can lead to unexpected behavior. Take for example the scenario where the PH5 api is being used in a server environment. Calling `sys.exit()` could crash the entire server.

Instead of calling `sys.exit()`, modules should be raising exceptions and catching them at a higher level.

derick-hess, 2019-02-07:
@damhuonglan as an example; in ph5_merge_helper.py all of the sys.exit() calls should eb removed and instead raise an exception and then those exceptions should be caught in main()

derick-hess, 2019-02-07:
also make sure not to add things like below to files.
```
################################################################
#
# modification
# version: 2019.032
# author: Lan Dam
# When multiple days are request, cut on day boundaries instead of
# every 24 hours
```

ascire-pic, 2020-10-23:
Closing this issue due to beginning development of GeoHDF

closed 2020-10-23 by ascire-pic