PIC-IRIS/PH5 #216: Commented out code should be removed

nick-iris, 2018-09-26:
Lots of the modules have commented out code that should be removed. This code will still be in source control once removed, so you are not gaining anything by keeping it in the code.

nick-iris, 2018-09-27:
@derick-hess Are you okay with me going through the codebase and removing old commented out code?

derick-hess, 2018-09-27:
Yeah that's a good idea.

nick-iris, 2018-09-27:
Okay I'll make these changes on the `cleanup` branch.

derick-hess, 2018-09-27:
okay thanks. I'l start addressing the other issues on monday morning when I get back in the office. I've been on a trip to Woods Hole this week

nick-iris, 2018-09-27:
Okay perfect. Perhaps review the changes I make on the `cleanup` branch first so that we don't run into merge conflicts.

PR #226 (2018-10-16): Cleaned up codebase and fixed miscellaneous issues
This PR addresses the following issues:

- Replaced all use of the deprecated optparse module with the argparse module
- Fix #224 - meta_data_gen now only reads what it needs to for the output.
- Fix #223 - improvements made to initialize_ph5.
- Fix #216 - Cleanup comments and remove commented out code.
- Fix #214 - 
  - All logging in modules has been updated to use a standard logging format defined in `~ph5.__init__.py`.
  - Add arguments to `ph5_total.py` to specify directory path and nickname. (e.g. master.ph5) The default behavior remains [... cut]
commit 790d1b27 (2018-10-16): Fix #216. Cleanup comments and remove commented out code.
closed 2018-10-16 by derick-hess