PIC-IRIS/PH5 #220: sort_kef_gen should have an option to write to a file

nick-iris, 2018-09-26:
Currently `sort_kef_gen.py` does not have an option to write to a file. By default it only writes to `stdout`, which blows up the terminal.

It should have something like this:
``` python
parser.add_argument("-o", "--outfile", action="store",
                    help=("KEF file write /Experiment_g/Sorts_g/Sort_t entries to. "
                               "Defaults to stdout."),
                    default=sys.stdout)
```

derick-hess, 2018-09-26:
Thanks . A few other tools do that and expect you to pipe the output to a file. I'll fix them all at once

nick-iris, 2018-09-26:
I have a major commit that I will push tomorrow. It includes a complete overhaul of the way logging is handled as well as some smaller fixes.

nick-iris, 2018-09-27:
`tabletokef` also need this option.

derick-hess, 2019-01-28:
@damhuonglan sort_kef_gen tabletokef and time_kef_gen. Currently output their kef files directly to stdout. Instead they need to write their output to a file.

I will get you a small  ph5 archive to test against and show you how they currently work.

closed 2019-02-05 by derick-hess