PIC-IRIS/PH5 #229: ph5_validate checks

nick-iris, 2018-10-04:
I just finished doing a thorough review of `ph5_validate`. Below I've created a list of items that go unchecked. Feel free to add to this if you think of any others.

The following should be added:

**Experiment_t Checks:**
- [ ] Check if there is any timeseries data outside of the  experiment start/end time span limits.
- [ ] Check if a report number matches the pattern ##-###. Currently there is only a check for its existence.
- [ ] Check if a SEED network code is between 1 and 2 characters. Currently there is only a check for its existence.
- [ ] Check for `summary_paragraph_s`.

**Array_t Checks:**
- [ ] Check for `description_s`
- [ ] Check if a `seed_location_code_s` between 0 and 2 characters. Currently there is NO check for `seed_location_code_s`.
- [ ] Check if a `seed_channel_code_s` is 3 characters. Currently there is NO check for `seed_channel_code_s`.
- [ ] Check for `location/Z/value_d`. Currently only location/X/value_d and location/Y/value_d are checked.
- [ ] Check for `das/serial_number_s`
- [ ] Check for `sensor/serial_number_s`

**Event_t Checks (if Event_t present)**
- [ ] Check for `id_s`
- [ ] Check for `description_s`
- [ ] Check for `location/coordinate_system_s`
- [ ] Check for `location/projection_s`
- [ ] Check for `location/ellipsoid_s`
- [ ] Check for `location/description_s`
- [ ] Check for `location/X/value_d`
- [ ] Check for `location/X/units_s`
- [ ] Check for `location/Y/value_d`
- [ ] Check for `location/Y/units_s`
- [ ] Check for `location/Z/value_d`
- [ ] Check for `location/Z/units_s`
- [ ] Check for `time/epoch_l`
- [ ] Check for `time/micro_seconds_i`
- [ ] Check for `size/value_d`
- [ ] Check for `size/units_s`
- [ ] Check for `depth/value_d`
- [ ] Check for `depth/units_s`

... Should we also have checks for Offset_t_XXX_XXX and Sort_t?

derick-hess, 2018-10-05:
For the offset table we should check if it exists if there are both shots and stations. If an offset table is missing ph5_validate can run geo_kef_gen to create the offset table for the user.

Sort table the same. Check if it exists and create one for the user if not.

I'll also edit your original post today with info on what should be errors and what should be warnings.

nick-iris, 2018-10-05:
Thanks. I've made a number of enhancements to ph5_validate that I'll push later today.

nick-iris, 2018-10-05:
I just pushed a update to the cleanup branch for `ph5_validate`. The update includes checks for all the items that I checked off the list in this issue. Additionally, the updated version writes to the log file only when there is  info/warning/error to report.

derick-hess, 2018-10-05:
Thanks. I will work on checking the time series data.

derick-hess, 2018-10-17:
working on the data check now

nick-iris, 2019-01-29:
@derick-hess I had a few other addtional features that I wrote and forgot to merge. I'll work on incorporating the with your changes. My updates included:

- Add ability to specify output file in ph5_validate.

For example:
`ph5_validate -n master.ph5 -p /hdf5-data2/PH5_Experiments/pn4/16-015/ -o 16-015_validate.log`
writes the validation output to a file called `16-015_validate.log`

- Add check to see if multiple experiment records exists in experiment_t
- Add warning logging level
- Move check for channels without data to the warning logging level.
- Improve error handling to avoid having users see a stack trace.

derick-hess, 2019-01-29:
okay sounds good

nick-iris, 2019-01-30:
Thanks I just realized that there were a few other checks that were accidentally lost when I rebased. I'm going to add the now before you merge.

nick-iris, 2019-01-30:
Oops, already merged. They were minor though, so I outlined them below for you to add if you wish:

In `check_experiment_t_completeness` add check to see that there is only one entry:
i.e.
```
        if len(experiment_t) > 1:
            error.append("More than one entry found in experiment_t.")
```

Errors caught in the `main()` method could probably all be written using the LOGGER.
i.e.
```
    try:
        ....
    except ph5api.APIError as e:
        LOGGER.error(e)
    except PH5ValidateException as e:
        LOGGER.error(e)
    except Exception as e:
        LOGGER.error(e)
```

derick-hess, 2019-01-30:
Okay I can add those in the morning before this gets merged into master

nick-iris, 2019-01-30:
I get a `KeyError` when running this against certain experiments.

i.e.
```
| => ph5_validate -n master.ph5 -p /hdf5-data2/PH5_Experiments/pn4/16-015 -o 16-015_validate.log --level=INFO
[2019-01-30 00:19:05,146] - ph5.utilities.ph5validate - INFO: Validating Experiment_t
[2019-01-30 00:19:05,147] - ph5.utilities.ph5validate - INFO: Validating Array_t
Traceback (most recent call last):
  File "/Users/nick/anaconda/envs/ph5/bin/ph5_validate", line 11, in <module>
    load_entry_point('ph5', 'console_scripts', 'ph5_validate')()
  File "/Users/nick/Development/workspace/PH5/ph5/utilities/ph5validate.py", line 753, in main
    validation_blocks.extend(ph5validate.check_array_t())
  File "/Users/nick/Development/workspace/PH5/ph5/utilities/ph5validate.py", line 548, in check_array_t
    self.check_station_completeness(station)
  File "/Users/nick/Development/workspace/PH5/ph5/utilities/ph5validate.py", line 504, in check_station_completeness
    component=channel_id)
  File "/Users/nick/Development/workspace/PH5/ph5/core/ph5api.py", line 1397, in get_extent
    Das_t = filter_das_t(self.Das_t[das]['rows'], component)
KeyError: '1X1057'
```

derick-hess, 2019-01-30:
@nick-iris Okay changes should be added and bug fixed.

commit 73d84aec (2018-10-05): Fix #229. Update ph5_validate to include more checks.
commit 88afdedc (2019-01-30): Merge pull request #252 from PIC-IRIS/ph5_validate

fixes #229
closed 2019-01-30 by derick-hess