PIC-IRIS/PH5 #264: ph5toms writes duplicate data

nick-iris, 2019-02-06:
**Describe the bug**
`ph5toms` is writing the same stream of data more than one time. This goes unnoticed when running on the command line since the same file gets overwritten on the second write. When running for the web service however, the same data is duplicated in the miniSEED output.

**Environment (please complete the following information):**
 - Program Version [ph5toms 2018.268]

**To Reproduce**
Steps to reproduce the behavior:

`
ph5toms -n master.ph5 -p /hdf5-data2/PH5_Experiments/pn4/14-032/ --starttime=2014-07-15T00:00:00.0 --stoptime=2014-07-29T00:00:00.0 --channel=DPZ --station=3205
`
writes 3 files:

```
XD.3205..DPZ.2014-07-24T035955.000000.ms
XD.3205..DPZ.2014-07-25T035955.000000.ms 
XD.3205..DPZ.2014-07-26T035955.000000.ms
```

These files contain no duplicates since they were overwritten after each write. But if you print the ObsPy `stream` in the source code before each write you will see that they are being duplicated.

---------------

Now see the same request through the web service:

http://service.iris.edu/ph5ws/dataselect/1/query?reqtype=FDSN&format=mseed&net=XD&sta=3205&cha=DPZ&starttime=2014-07-15&endtime=2014-07-29&nodata=404

```
msi -T -Q ph5ws-dataselect_2019-02-06t18_31_49z.mseed
  Source                Start sample             End sample        Hz  Samples
XD_3205__DPZ_D    2014,205,03:59:55.000000 2014,205,13:59:54.996000 250 9000000
XD_3205__DPZ_D    2014,205,03:59:55.000000 2014,205,13:59:54.996000 250 9000000
XD_3205__DPZ_D    2014,206,03:59:55.000000 2014,206,13:59:54.996000 250 9000000
XD_3205__DPZ_D    2014,206,03:59:55.000000 2014,206,13:59:54.996000 250 9000000
XD_3205__DPZ_D    2014,207,03:59:55.000000 2014,207,08:29:54.996000 250 4050000
XD_3205__DPZ_D    2014,207,03:59:55.000000 2014,207,08:29:54.996000 250 4050000
```

You see here that the data was clearly duplicated. The web service sees the issue since data cannot be overwritten as it is streamed as soon as it is loaded.

**Expected behavior**
Data should not be duplicated in the output.

**Additional context**
We discovered this problem while doing testing for enabling PH5 web services to work with BREQ_FAST request.

nick-iris, 2019-02-06:
I think that since a single station belongs to 2 arrays it is being read twice.

Printing the stations cuts that get written shows the following:

```
{'array_code': '001', 'elev': 1160.18798828, 'sensor_type': 'Geospace GS11V', 'das_model': 'rt125', 'receiver_n_i': 0, 'station': '3205', 'das_manufacturer': 'Reftek', 'location': '', 'latitude': 46.2151689269, 'net_code': 'XD', 'sample_rate_multiplier': 1, 'notimecorrect': False, 'component': 1, 'das': '10714', 'endtime': 1406246400, 'response_n_i': 1, 'seed_channel': 'DPZ', 'longitude': -122.243570164, 'sample_rate': 250, 'experiment_id': '14-032', 'starttime': 1406171341, 'seed_station': '3205'}
{'array_code': '001', 'elev': 1160.18798828, 'sensor_type': 'Geospace GS11V', 'das_model': 'rt125', 'receiver_n_i': 0, 'station': '3205', 'das_manufacturer': 'Reftek', 'location': '', 'latitude': 46.2151689269, 'net_code': 'XD', 'sample_rate_multiplier': 1, 'notimecorrect': False, 'component': 1, 'das': '10714', 'endtime': 1406332800, 'response_n_i': 1, 'seed_channel': 'DPZ', 'longitude': -122.243570164, 'sample_rate': 250, 'experiment_id': '14-032', 'starttime': 1406246400, 'seed_station': '3205'}
{'array_code': '001', 'elev': 1160.18798828, 'sensor_type': 'Geospace GS11V', 'das_model': 'rt125', 'receiver_n_i': 0, 'station': '3205', 'das_manufacturer': 'Reftek', 'location': '', 'latitude': 46.2151689269, 'net_code': 'XD', 'sample_rate_multiplier': 1, 'notimecorrect': False, 'component': 1, 'das': '10714', 'endtime': 1406419200, 'response_n_i': 1, 'seed_channel': 'DPZ', 'longitude': -122.243570164, 'sample_rate': 250, 'experiment_id': '14-032', 'starttime': 1406332800, 'seed_station': '3205'}
{'array_code': '060', 'elev': 1160.18798828, 'sensor_type': 'Geospace GS11V', 'das_model': 'rt125', 'receiver_n_i': 0, 'station': '3205', 'das_manufacturer': 'Reftek', 'location': '', 'latitude': 46.2151689269, 'net_code': 'XD', 'sample_rate_multiplier': 1, 'notimecorrect': False, 'component': 1, 'das': '10714', 'endtime': 1406246400, 'response_n_i': 1, 'seed_channel': 'DPZ', 'longitude': -122.243570164, 'sample_rate': 250, 'experiment_id': '14-032', 'starttime': 1406171341, 'seed_station': '3205'}
{'array_code': '060', 'elev': 1160.18798828, 'sensor_type': 'Geospace GS11V', 'das_model': 'rt125', 'receiver_n_i': 0, 'station': '3205', 'das_manufacturer': 'Reftek', 'location': '', 'latitude': 46.2151689269, 'net_code': 'XD', 'sample_rate_multiplier': 1, 'notimecorrect': False, 'component': 1, 'das': '10714', 'endtime': 1406332800, 'response_n_i': 1, 'seed_channel': 'DPZ', 'longitude': -122.243570164, 'sample_rate': 250, 'experiment_id': '14-032', 'starttime': 1406246400, 'seed_station': '3205'}
{'array_code': '060', 'elev': 1160.18798828, 'sensor_type': 'Geospace GS11V', 'das_model': 'rt125', 'receiver_n_i': 0, 'station': '3205', 'das_manufacturer': 'Reftek', 'location': '', 'latitude': 46.2151689269, 'net_code': 'XD', 'sample_rate_multiplier': 1, 'notimecorrect': False, 'component': 1, 'das': '10714', 'endtime': 1406419200, 'response_n_i': 1, 'seed_channel': 'DPZ', 'longitude': -122.243570164, 'sample_rate': 250, 'experiment_id': '14-032', 'starttime': 1406332800, 'seed_station': '3205'}
```

derick-hess, 2019-02-06:
That looks like a problem in the actual experiment. I'll take a look at the experiment.  The only reason that a station should be duplicated in another array is if it has a different sample rate and in that case it would be two different data streams.

We can attempt to fix this in PH5toms but I thin it would slow down processing. I think it might be better to fix this experiment and add an handling an an error in PH5validate. Making it so exactly duplicated stations would cause ph5validate to fail the archive as not ready for the DMC.

derick-hess, 2019-02-06:
Since none of the equipment changed and neither did the location for that station possibly being deployed 3 times then there should be only one entry in the array table for it covering the entire time span.

derick-hess, 2019-02-06:
I'm looking at this now and testing and you are correct the issue comes from the exact same metadata being in both array 001 and 060.  This can definitely be solved by adding a test in ph5validate.

derick-hess, 2019-02-06:
I'll propose a change to ph5validate that fixes this issue. I think hashing each station metadata and comparing would be an efficient way to check if the exact same metadata exists in multiple places

derick-hess, 2019-02-06:
Proposed fix to ph5toms:

Before yielding station cut object in create a hash of the (station, das, channel, lat, lon, sample rate, and sample rate multiplier) .

Check this hash against a list of previously added hashes. If there is a match do not yield station_cut object. If there is not a match add the has to the list.

nick-iris, 2019-02-06:
I think that will work. These are the edge cases that it would be nice to write unit test for if we ever create a small test dataset.

commit 0b37d3e1 (2019-02-07): fixes #264
commit 394f81b8 (2019-02-07): Merge pull request #268 from PIC-IRIS/ph5toms

Fixes #264
closed 2019-02-07 by derick-hess