PIC-IRIS/PH5 #273: Streaming SEG-Y for PH5 web services

nick-iris, 2019-02-16:
**Is your feature request related to a problem? Please describe.**
Currently the PH5 web services do not provide a way to request a stream of SEG-Y data.

**Describe the solution you'd like**
 If we can calculate the expected number of SEG-Y data traces more accurately using the data availability methods, then we will no longer need to update the reel headers after the SEG-Y file has been written. This would make it possible to stream SEG-Y directly from the PH5 dataselect web service.

**Additional context**
If I'm not mistaking, the following code is preventing us from being able to stream SEG-Y from the PH5 dataselect web service.

ph5toevt: https://github.com/PIC-IRIS/PH5/blob/master/ph5/clients/ph5toevt.py#L585-L600
ph5torec: https://github.com/PIC-IRIS/PH5/blob/master/ph5/clients/ph5torec.py#L505-L515

We are having to go back and update the reel header since the estimated number of traces was inaccurate. I'm proposing that we use the data availability information to accurately predetermine the number of traces that will be written. We should then be able to stream SEG-Y directly to users.

derick-hess, 2019-02-16:
I think this could work. For speed sake should we hjave a setting to check the tsindex database if it exists and if not call the availability methods? This way it will work from the command line and be faster for webservices if tsindex table exists.

nick-iris, 2019-03-05:
One issue is that the `segyfactory.Ssegy` class relies on PH5 table instances to be set. By the time a ObsPy trace object is returned from ph5toms the web service program knows nothing about the internals of PH5.

I'd like to write a new version of segyfactory that works with the ObsPy traces that are returned from ph5toms. In this new version, the `write_segy_hdr`, `set_trace_header`, `set_ext_header_seg`, ... etc methods would accept each of their fields directly as input parameters instead of reading them from the PH5 tables. The advantage to this would be that the API could be used by anyone wanting to write SEG-Y and not just PH5.

Do you see any reason why this wouldn't work?

derick-hess, 2019-03-05:
I don't see any reason why this wouldn't work. I've been doing a lot int eh opposite direction. Taking obspy streams and putting them in to PH5 and it works well. Header information should be easy enough and I think the data arrays used to write out the traces are already numpy arrays so there should be no problem feeding obspy trace data to a modified segyfactory as well.

nick-iris, 2019-11-22:
SEGY streaming capability was added to the ph5 dataselect service code.

closed 2019-11-22 by nick-iris