PIC-IRIS/PH5 #280: ph5toevt request results in unexpected error

nick-iris, 2019-03-08:
**Describe the bug**
I made a `ph5toevt` request that resulted in an unexpected error.

**To Reproduce**
`ph5toevt --path=/hdf5-data2/PH5_Experiments/pn4/15-016/ --nickname=master.ph5 --length=120 -A=001 --offset=0 --reduction_velocity=0 --eventnumber=5013 --shot_line=001`

derick-hess, 2019-03-08:
Thanks i'll fix this. That experiment must not have an offset table, but an offset table also shouldn't be required. I will give you a kef to fix that experiment, then update experiment to not fail adn return an empty table if it doesn't exist.

derick-hess, 2019-03-08:
I'll also add a feature request to ph5_validate to check for this. IF there are shot lines then the offset table should be calculated and present in PH5

derick-hess, 2019-03-08:
I have a fix. Running tests on all the data sets I have in house that have events to make sure. Should have the fix posted within an hour

closed 2019-03-12 by derick-hess