PIC-IRIS/PH5 #282: Unit Tests Needed

derick-hess, 2019-03-15:
Now that building a test data set has been added to PH5, more unit tests can be developed. 
I am going to add a checklist below to keep track of what packages have unit tests.  Refer to this issue when adding a PR that adds unit tests and check off the package that you have added unit tests for.  When unit tests have been created for all packages this issue can be closed. 

Please don't check off a package until the unit tests cover as much of the package as possible.

Clients
- [ ] ph5availability
- [ ] ph5toevt
- [ ] ph5torec
- [ ] ph5toexml
- [ ] ph5toms
- [ ] ph5tostationxml

Core
- [ ] columns
- [ ] cs2cs
- [ ] decimate
- [ ] ebcdic
- [ ] experiment
- [ ] external_file
- [ ] ibmfloat
- [ ] kef
- [ ] kefutility
- [ ] kefx
- [ ] mplturkey
- [x] ph5api
- [ ] ph5utils
- [ ] pmonitor
- [ ] pn125
- [ ] pn130
- [ ] rt_130_h
- [ ] sac_h (maybe obsolete and need removal)
- [ ] sacfactory (maybe obsolete and need removal)
- [ ] sacreader (maybe obsolete and need removal)
- [ ] segd_h
- [ ] segdreader
- [ ] segy_h
- [ ] segyfactory
- [ ] segyreader
- [ ] sfactory
- [ ] timedoy

Utilities
- [ ] 125a2ph5
- [ ] 1302ph5
- [ ] changes
- [ ] cross_check_event_array_data
- [ ] csvtokef
- [ ] dumpfair
- [ ] dumpsac (Needs rewrite...use Obspy)
- [ ] dumpsgy
- [ ] fix_3chan_texan
- [ ] fix_num_samples
- [ ] geod2kef
- [ ] grao2ph5 (obsolete...remove)
- [ ] index_offset_t
- [ ] initialize_ph5
- [ ] kef2kml
- [ ] kef2ph5
- [ ] kefedit
- [ ] keftocsv
- [ ] load_das_t
- [ ] meta_data_gen
- [X] metadatatoph5
- [ ] noven
- [ ] novenkef
- [ ] novenqc
- [ ] novitiate (might be obsolete)
- [ ] nuke_tables
- [x] obspytoph5
- [ ] pforma_io
- [ ] pformacl
- [ ] pformagui
- [ ] ph5_help
- [ ] ph5_merge_helper
- [ ] ph5_total
- [ ] ph5validate
- [ ] segd_h
- [ ] recreate_external_references
- [ ] report2ph5
- [ ] report_gen
- [ ] resp_load
- [ ] seg2toph5
- [ ] segd2ph5
- [ ] segy2ph5
- [ ] set_deploy_pickup_times
- [ ] set_n_i_response
- [ ] sort_array_t
- [ ] sort_kef_gen
- [ ] tabletokef
- [ ] time_kef_gen
- [ ] unsimpleton
- [ ] watchit

derick-hess, 2019-03-22:
@damhuonglan @dthomas1953 I think our highest priority should be the clients, ph5toms especially, and then ___toph5 programs.

damhuonglan, 2019-04-12:
Anyone who are working unit test for a package should comment it on this issue so our works don't overlap each other. 
I currently working on ph5availability's unit test

ascire-pic, 2020-10-23:
Appropriate unit tests are being created as changes are being implemented. As a result, I am closing this issue since unit tests are no longer being implemented as described in this issue.

closed 2020-10-23 by ascire-pic