PIC-IRIS/PH5 #289: [FEATURE] PH5 Magnetotellurics Metadata

derick-hess, 2019-04-03:
**Is your feature request related to a problem? Please describe.**
Add new metadata fields to capture magnetotelluric metadata not currently captured

channel:
Declination
Declination_epoch

response:
Contact_resistance
Length

Provenance table (for seismic and mt):
Creating_application
Creation_time

Creator
Email
Name
Organization
Organization_url

Submitter
Email
Name
Organization
organization_url

Software
Author
Email
Name
Organization
Organization_url
Name
Version
Release_date

.

derick-hess, 2019-04-03:
I'll format the above PR for easier viewing

timronan, 2020-12-21:
This is being handled by  MTH5 https://mth5.readthedocs.io/en/tables/index.html and a different group.

closed 2020-12-21 by timronan