PIC-IRIS/PH5 #292: [BUG] VM1 and VM2 channels missing and VM3 channels have many gaps

nick-iris, 2019-04-04:
**Describe the bug**
There is an issue that we found for YW.513. This station has VM1,VM3,VM3,HH1,HH2,HHZ channels in PH5 but a request for all channels shows only VM3:

http://service.iris.edu/ph5wsbeta/availability/1/query?format=text&net=YW&sta=513&loc=*&cha=*&starttime=2016-06-01&endtime=2016-08-01&mergequality=false&mergesamplerate=false&mergeoverlap=false&showlastupdate=false&excludetoolarge=true&nodata=404

#n s     l  c   q sample-rate                    earliest                      latest
YW 513   -- VM3 D         1.0 2016-06-24T22:30:58.640000Z 2016-06-24T22:33:55.640000Z
YW 513   -- VM3 D         1.0 2016-06-24T22:34:25.640000Z 2016-07-20T13:57:55.640000Z
YW 513   -- VM3 D         1.0 2016-07-20T14:02:20.640000Z 2016-08-01T00:00:00.000000Z
In addition, the sample rate for VM3 is 0.1 Hz, not 1 Hz as reported by the availability service. The actual data I get back from dataselect has many gaps that are not accounted for in the availability service (even considering the error in the sample rate):

> msi -T  ph5ws-dataselect_2019-04-04t01_44_39z.mseed
   Source                Start sample             End sample        Hz  Samples
YW_513__VM3       2016,176,22:30:00.000000 2016,176,22:39:50.000000 0.1 60
YW_513__VM3       2016,176,22:30:00.000000 2016,176,22:39:50.000000 0.1 60
YW_513__VM3       2016,177,04:18:10.000000 2016,177,04:29:50.000000 0.1 71
YW_513__VM3       2016,177,04:18:10.000000 2016,177,04:28:00.000000 0.1 60
YW_513__VM3       2016,177,04:18:10.000000 2016,177,04:19:50.000000 0.1 11
YW_513__VM3       2016,177,10:06:30.000000 2016,177,10:09:50.000000 0.1 21
YW_513__VM3       2016,177,10:06:30.000000 2016,177,10:09:50.000000 0.1 21
YW_513__VM3       2016,177,15:54:50.000000 2016,177,15:59:50.000000 0.1 31
YW_513__VM3       2016,177,15:54:50.000000 2016,177,15:59:50.000000 0.1 31
YW_513__VM3       2016,177,21:43:10.000000 2016,177,21:49:50.000000 0.1 41
YW_513__VM3       2016,177,21:43:10.000000 2016,177,21:49:50.000000 0.1 41
YW_513__VM3       2016,178,03:31:30.000000 2016,178,03:39:50.000000 0.1 51
YW_513__VM3       2016,178,09:19:50.000000 2016,178,09:29:50.000000 0.1 61
YW_513__VM3       2016,178,15:08:10.000000 2016,178,15:19:50.000000 0.1 71
YW_513__VM3       2016,178,20:56:30.000000 2016,178,20:59:50.000000 0.1 21
YW_513__VM3       2016,179,02:44:50.000000 2016,179,02:49:50.000000 0.1 31
YW_513__VM3       2016,179,08:33:10.000000 2016,179,08:39:50.000000 0.1 41
YW_513__VM3       2016,179,14:21:30.000000 2016,179,14:29:50.000000 0.1 51
YW_513__VM3       2016,179,20:09:50.000000 2016,179,20:19:50.000000 0.1 61
YW_513__VM3       2016,180,01:58:10.000000 2016,180,02:09:50.000000 0.1 71
YW_513__VM3       2016,180,07:46:30.000000 2016,180,07:49:50.000000 0.1 21
YW_513__VM3       2016,180,13:34:50.000000 2016,180,13:39:50.000000 0.1 31
YW_513__VM3       2016,180,19:23:10.000000 2016,180,19:29:50.000000 0.1 41
YW_513__VM3       2016,181,01:11:30.000000 2016,181,01:19:50.000000 0.1 51
YW_513__VM3       2016,181,06:59:50.000000 2016,181,07:09:50.000000 0.1 61
YW_513__VM3       2016,181,12:48:10.000000 2016,181,12:59:50.000000 0.1 71
YW_513__VM3       2016,181,18:36:30.000000 2016,181,18:39:50.000000 0.1 21
YW_513__VM3       2016,182,00:24:50.000000 2016,182,00:29:50.000000 0.1 31
YW_513__VM3       2016,182,06:13:10.000000 2016,182,06:19:50.000000 0.1 41
YW_513__VM3       2016,182,12:01:30.000000 2016,182,12:09:50.000000 0.1 51
YW_513__VM3       2016,182,17:49:50.000000 2016,182,17:59:50.000000 0.1 61
YW_513__VM3       2016,182,23:38:10.000000 2016,182,23:49:50.000000 0.1 71
YW_513__VM3       2016,183,05:26:30.000000 2016,183,05:29:50.000000 0.1 21
YW_513__VM3       2016,183,11:14:50.000000 2016,183,11:19:50.000000 0.1 31
YW_513__VM3       2016,183,17:03:10.000000 2016,183,17:09:50.000000 0.1 41
YW_513__VM3       2016,183,22:51:30.000000 2016,183,22:59:50.000000 0.1 51
YW_513__VM3       2016,184,04:39:50.000000 2016,184,04:49:50.000000 0.1 61
YW_513__VM3       2016,184,10:28:10.000000 2016,184,10:39:50.000000 0.1 71
YW_513__VM3       2016,184,16:16:30.000000 2016,184,16:19:50.000000 0.1 21
YW_513__VM3       2016,184,22:04:50.000000 2016,184,22:09:50.000000 0.1 31
YW_513__VM3       2016,185,03:53:10.000000 2016,185,03:59:50.000000 0.1 41
YW_513__VM3       2016,185,09:41:30.000000 2016,185,09:49:50.000000 0.1 51
YW_513__VM3       2016,185,15:29:50.000000 2016,185,15:39:50.000000 0.1 61
YW_513__VM3       2016,185,21:18:10.000000 2016,185,21:29:50.000000 0.1 71
YW_513__VM3       2016,186,03:06:30.000000 2016,186,03:09:50.000000 0.1 21
YW_513__VM3       2016,186,08:54:50.000000 2016,186,08:59:50.000000 0.1 31
YW_513__VM3       2016,186,14:43:10.000000 2016,186,14:49:50.000000 0.1 41
YW_513__VM3       2016,186,20:31:30.000000 2016,186,20:39:50.000000 0.1 51
YW_513__VM3       2016,187,02:19:50.000000 2016,187,02:29:50.000000 0.1 61
YW_513__VM3       2016,187,08:08:10.000000 2016,187,08:19:50.000000 0.1 71
YW_513__VM3       2016,187,13:56:30.000000 2016,187,13:59:50.000000 0.1 21
YW_513__VM3       2016,187,19:44:50.000000 2016,187,19:49:50.000000 0.1 31
YW_513__VM3       2016,188,01:33:10.000000 2016,188,01:39:50.000000 0.1 41
YW_513__VM3       2016,188,07:21:30.000000 2016,188,07:29:50.000000 0.1 51
YW_513__VM3       2016,188,13:09:50.000000 2016,188,13:19:50.000000 0.1 61
YW_513__VM3       2016,188,18:58:10.000000 2016,188,19:09:50.000000 0.1 71
YW_513__VM3       2016,189,00:46:30.000000 2016,189,00:49:50.000000 0.1 21
[... 43 lines cut ...]
YW_513__VM3       2016,199,16:13:10.000000 2016,199,16:19:50.000000 0.1 41
YW_513__VM3       2016,199,22:01:30.000000 2016,199,22:09:50.000000 0.1 51
YW_513__VM3       2016,200,03:49:50.000000 2016,200,03:59:50.000000 0.1 61
YW_513__VM3       2016,200,09:38:10.000000 2016,200,09:49:50.000000 0.1 71
YW_513__VM3       2016,200,15:26:30.000000 2016,200,15:29:50.000000 0.1 21
YW_513__VM3       2016,200,21:14:50.000000 2016,200,21:19:50.000000 0.1 31
YW_513__VM3       2016,201,03:03:10.000000 2016,201,03:09:50.000000 0.1 41
YW_513__VM3       2016,201,08:51:30.000000 2016,201,08:59:50.000000 0.1 51
YW_513__VM3       2016,201,14:39:50.000000 2016,201,14:49:50.000000 0.1 61
YW_513__VM3       2016,201,20:28:10.000000 2016,201,20:39:50.000000 0.1 71
YW_513__VM3       2016,202,02:16:30.000000 2016,202,02:19:50.000000 0.1 21
YW_513__VM3       2016,202,08:04:50.000000 2016,202,08:09:50.000000 0.1 31
YW_513__VM3       2016,202,13:53:10.000000 2016,202,13:58:00.000000 0.1 30
YW_513__VM3       2016,202,14:01:30.000000 2016,202,14:09:50.000000 0.1 51
YW_513__VM3       2016,202,19:49:50.000000 2016,202,19:59:50.000000 0.1 61
Total: 1 trace(s) with 115 segment(s)

The data in our primary archive does not have these gaps:

> msi -T -M 'VM3' 17[6-9]/513* 1[89]*/513* 20[12]/513*
   Source                Start sample             End sample        Hz  Samples
YW_513__VM3       2016,176,22:30:00.000000 2016,199,23:59:50.000000 0.1 199260
YW_513__VM3       2016,201,00:00:00.000000 2016,202,13:58:00.000000 0.1 13669
YW_513__VM3       2016,202,14:01:30.000000 2016,202,23:59:50.000000 0.1 3591
Total: 1 trace(s) with 3 segment(s)

derick-hess, 2019-04-04:
I'll take a look at the raw data to see if the bug is coming from writing the data out or if it wasn't fully loaded originally

derick-hess, 2019-04-04:
Also thinking about it...I think there was an issue with VM data  from rt130s in a later experiment that Steve may have been fixed. I'll need to check the data. It may be the case that the data needs to be reloaded.

nick-iris, 2019-04-04:
Thanks Derick. Pinging @gillian-iris since she uncovered this issue.

derick-hess, 2019-04-04:
Looking at the das table it sees the following for VM3:

June 24 22:30:00 sample rate 0.1 samples 2090
June 25 04:18:00 sample rate 0.1 samples 2090

So it looks like it is a data loading issue. I'll take a look at the raw data and see what needs to be done.

derick-hess, 2019-04-04:
I rebuilt from the raw data and things look way better. The actual data arrays are okay in the original wavefields but the das table is wrong for the vm channels. When I just plotted the raw data with matplotlib it looks fine but the bad metadata makes all those gaps.

When I reloaded the raw data for 513 with the latest PH5 code everything looks like I expect and trace start stop times match what you show above and what rt2ms outputs.

I will try to log in to the server at the PIC when I get to Vienna and fix Wavefields (along with adding the LOG channel for that one station).. r if it isn't super urgent I can wait until I get back when hoepfully the new rt2ms code is written and then I can create miniseed log channels to add to wavefields of all the broadband data

nick-iris, 2019-04-04:
Thanks Derick. I think you can wait until you get back from EGU to fix this and then load the LOG channels as well.

The HH* channels were also missing from station 513. Can you also include these as well when you reload the data?

gillian-iris, 2019-04-04:
The HH* channels are missing from the availability service return, but I believe the data is there in the PH5 archive.

-Gillian

derick-hess, 2019-04-04:
Thanks, yeah I'll take a look at the availability as well. I suspect when it is grabbing the trace metadata it isn't fully filtering on sample rate correctly.

nick-iris, 2019-04-16:
@derick-hess Instead of updating Wavefields directly on the server you should send us the changes and the data group at the DMC will modify the archive. This will ensure that there are procedures for updates  similar to this one in place and that operations has a record of exactly what changed.

derick-hess, 2019-04-17:
@nick-iris I'll work on doing it that way. Right now i'm simultaneously updating the old one I have here at the PIC and recreating wavefields from scratch as well. Recreating it is mostly just for testing purposes of the new processing machines we have here, but I'll also compare the two as well to make sure no differences exist between the fully recreated from scratch and the updated version

nick-iris, 2019-06-17:
@derick-hess Alissa located Wavefields folders where you were working on this fix. What exactly where you planning on sending us? Was the plan to replace the entire PH5, or is there a subset of the PH5 that just needs to be replaced? Thanks for the help.

nick-iris, 2019-06-25:
@derick-hess Can you please provide us with some background on what you changed in Wavefields experiment? Do we need to replace the entire experiment at the DMC, or just a subset of the PH5?

ascire-pic, 2020-10-23:
Wavefields (YW) PH5 has been replaced with an updated version in July 2019. As a result the VM* and HH* channels are now correct.

closed 2020-10-23 by ascire-pic