PIC-IRIS/PH5 #293: [BUG] YW LH* channels not returned from PH5

nick-iris, 2019-04-04:
**Describe the bug**
We also have YW LH1,LH2,LHZ channels in our primary archive, but requesting those from PH5 gives a ‘No Content’ message. I don’t know if they are supposed to be in the PH5 archive.

derick-hess, 2019-04-04:
I'll check, but if I remember correctly the data is there but for some reason the data group didn't create array tables for these.

derick-hess, 2019-04-04:
I can confirm the LH* data is definitely present, just no array tables exist to expose it. I don't see any reason why it shouldn't be exposed so I'll confirm with the data group

nick-iris, 2019-04-04:
Sounds good, let me know what they say. Thanks for looking into this.

ascire-pic, 2020-10-23:
Wavefields (YW) PH5 has been replaced with an updated version in July 2019. As a result, the LH* channels are now accessible as they have the correct metadata.

closed 2020-10-23 by ascire-pic