PIC-IRIS/PH5 #296: Construct should be updated to version 2.9

nick-iris, 2019-05-24:
The version of the `construct` library that is in PH5 uses for writing SEG-Y is very old and should be updated. I am unable to find documentation for version 2.5 that is currently used in PH5.

More information is available at https://construct.readthedocs.io/en/latest/transition29.html.

derick-hess, 2019-05-24:
I'll have to check, but I remember Steve pinned 2.5 because newer version of construct didn't work with whatever he had written.

derick-hess, 2019-05-24:
I probably won't be able to fix it before I leave, but I'll do some digging and find out the issues and post them here. It's definitely something that needs to be fixed and updated. Version 2.5 is very old.

nick-iris, 2019-05-24:
There isn't any huge rush for this. Just wanted to have it documented.

nick-iris, 2020-01-29:
@dsentinel I'm going to work on updating this in PH5 as it's causing conflicts with the web service deployment which uses a new version of the construct library

nick-iris, 2020-01-29:
The PH5 web services use construct version 2.9.45

nick-iris, 2020-01-29:
I already have code for writing the SEG-Y that the web services produce.

Can we just use this for the PH5 library or do we need to support the PASSCAL, Menlo Park, and iNova extended trace headers?

There may be an advantage in remaining consistent between what the web service produces and what gets produced by the PH5 library. It bet many of these formats were added only to please a handful of PIs at one time or another.

dsentinel, 2020-01-30:
I’ve CC’ Bruce and the data group.

Is this rev2, 1 or both?

If memory serves the PASSCAL SEGY is for single trace per file just trace header, this is old and no longer supported/needed.

There _may_ be a PASSCAL SEGY gather which allows for longer trace lengths, but I think this was alleviated with rev2.
We do have this https://www.passcal.nmt.edu/content/seg-y-files-produced-ph5-data-sets
Only 2 bytes for Number of samples. This looks standard. Is it different from classic SEGY?

I don’t know much about Menlo, and iNova, maybe others can comment, and correct any errors above.

Cheers,
Lloyd

nick-iris, 2020-01-30:
Thanks Lloyd. It's not hard to keep them if we choose, just thought it might easier to maintain without so many variations.

dsentinel, 2020-01-31:
I agree, happy to drop them if we have no use for them.