PIC-IRIS/PH5 #297: New Help Methodology - comments solicited

dthomas1953, 2019-05-29:
I'm trying out a new methodology for the help utility.

A working model is online at 
https://github.com/PIC-IRIS/PH5/tree/newhelp

The old methodology included a bunch of canned descriptions for scripts/clients and such. As these were divorced from their source files, it's hard to maintain.

For the new approach, I'm adding a class definition for each module. Only 2 have been added in this draft version, in ph5.clients.ph5view.ph5_viewer.py, and in ph5.clients.ph5toevt.py. 

Then, in ph5.utilities.ph5_help.py, if either of these two is included as the argument for ph5help, e.g.
(ph5) $ph5 ph5toevt
then the help file imports the given file, and grabs the description from the new class.

Here's how it's working so far:
```python
(ph5) spike:PH5 dthomas$ ph5 ph5toevt
ph5toevt: Generate SEG-Y gathers in shot order...

(ph5) spike:PH5 dthomas$ ph5 ph5viewer

Usage: ph5 name {subprogram name}
The name 'ph5viewer' is not supported.
Type ph5 with no arguments to see the full list.
ph5viewer:

(ph5) spike:PH5 dthomas$ ph5 ph5view
/Users/dthomas/miniconda/envs/ph5/lib/python2.7/site-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
ph5view:
ph5view is a GUI program for plotting responses,
saving files to SEGY, etc.
It has three main panels: Control, Main, and Support.
Control Panel: the panel with the menu and three tabs
(Control, Shot Gather, Receiver Gather).
Main Window: the main panel for plotting.
Support Window: the panel to show small portion of the
plot (the idea is to reduce the amount of trace need
to be analyzed).
Consult the Help Utility in ph5view to learn more.
```
Does this seem like a viable approach? I'm hoping for confirmation before I add description classes to all the scripts.

Thanks, Dave

dthomas1953, 2019-05-29:
Any ideas on how to suppress the 'Matplotlib' warning? It's time-consuming and annoying.

nick-iris, 2019-06-03:
@dthomas1953 I'll look this over now. Sorry for not getting to this sooner, it has been pretty busy here.

nick-iris, 2019-06-03:
@dthomas1953  I think we should abandon having the user type `ph5` as entry point for help documentation. Although I think your new approach could work, I think the overall usage of typing `ph5 <command>` in order to get help documentation is inconsistent with how users run other commands in the project.

I'm leaning towards keeping this simple. When a user types `ph5` a high level list of commands should get listed. If they want to see help documentation on a command they would just enter the command name with the `--help` option. A more detailed description of the approach I'd like to take is in issue https://github.com/PIC-IRIS/PH5/issues/294.

angstymeat, 2019-06-03:
@nick-iris The method you're describing, using `ph5` for a list of commands, and `ph5 command --help`, is a more standardized way of dealing with help and subcommands.

Something that might be of use is the [Click Toolkit](http://click.palletsprojects.com/en/7.x/) which provides a standard means of defining and accessing cli arguments, and will automatically generate help.

dthomas1953, 2019-06-03:
Looking at your ideas in #294 , will comment over there.

dthomas1953, 2019-06-18:
I closed it because it's the same issue as #294 .

closed 2019-06-25 by nick-iris