PIC-IRIS/PH5 #300: ph5toms - write log file on demand

ascire-pic, 2019-06-12:
For troubleshooting, ph5toms could include a flag that would write a log file

ph5toms used to write a log file by default, but that was removed when web services was implemented. However, that log file was very useful for determining why ph5toms failed to write data so having the option for a verbose log when needed would be helpful.

timronan, 2020-12-21:
At @ascire-pic do you know approximately when the log file function was disabled? Maybe we could use part of the old code. This could point to potential issues in the data format.

ascire-pic, 2020-12-23:
@timronan I think it was disabled when miniseed output was added to web services at the DMC. It used to default to writing a log file, either to a text file or to the screen (can't recall right now) and that was removed to accommodate writing out miniseed via web services to keep the DMC from ending up with endless log text files every time someone requested data. I am interested in having the option to write a log file because it was useful for troubleshooting when a user is having issues writing out miniseed data from a local PH5 archive.

timronan, 2020-12-23:
@ascire-pic it seems like a good idea to have a verbosity flag that will write a log file when executing ph5toms. It seems like this flag will help both users and developers trouble shoot ph5toms.