PIC-IRIS/PH5 #301: read UTM or UTM to lat/lon converter

ascire-pic, 2019-06-12:
Ability to read in UTM directly via noven or an accessible UTM converter to calculate lat/lon for metadata

The Fairfield nodes require location metadata in UTM. Currently pforma converts the UTM from the Fairfield continuous files to lat/lon to populate the array tables. However, if we need to update the array tables after that, frequently the location information is supplied in UTM since that is the format required by the node server.

PH5 being able to read in UTM in addition to lat/lon for location information or having an integrated UTM converter would help make metadata fixes for node locations easier.

El-minadero, 2019-06-20:
[pyproj](https://pypi.org/project/pyproj/) might be a good tool to leverage towards this end.

dthomas1953, 2019-06-20:
I've worked with UTM conversions quite a bit, will look at taking this project on.

dthomas1953, 2019-06-21:
I talked with George, who suggested a command line script capable of converting a file with UTM coords into lats/longs might be the way to go. You could run this on a .csv file with known UTM easting/northings/etc., and create a file with the corresponding lats and longs. Or, we could write the lats/longs directly into the relevant .kef file.  How do you want this to work?

Here's the output of a one-page converter, showing the original UTM file and the converted lat/long file.

```
$ cat utmdata.txt
easting, northing, zone, hemisphere
540947,1361594,58,S
322916,3771967,13,N

$ python utm2coords.py utmdata.txt latlong.out
converting utmdata.txt ==> latlong.out
***************************************
easting= 540947.0 , northing= 1361594.0 , zone= 58 , hemisphere= S
Lon, Lat =  166.73816638798527 -77.81567398301094
***************************************
easting= 322916.0 , northing= 3771967.0 , zone= 13 , hemisphere= N
Lon, Lat =  -106.91909595147378 34.07349577107704
***************************************

$ cat latlong.out
166.73816638798527,-77.81567398301094
-106.91909595147378,34.07349577107704
```

ascire-pic, 2020-10-23:
Current plan is to integrate this utility into PASSOFT instead of into PH5. As a result, will close this issue since it is no longer relevant to PH5.

closed 2020-10-23 by ascire-pic