PIC-IRIS/PH5 #302: [FEATURE] optionally have ph5_validate create experiment_t_bounding_box.kef

nick-iris, 2019-06-12:
**Is your feature request related to a problem? Please describe.**
Currently a file called experiment_t_bounding_box.kef is automatically created if this information is missing from a PH5 experiment. Often you want to just validate and not create this file, so I think this should be optional.

**Describe the solution you'd like**
Make the creation of the experiment_t_bounding_box.kef file optional by providing a command line option to ph5_validate.

dsentinel, 2019-07-30:
Is this addressed in #314 ?

nick-iris, 2019-07-30:
No this issue relates to the experiment_t_bounding_box.kef file that ph5_validate creates automatically. There needs to be a flag to tell ph5_validate to automatically create the file or not.