PIC-IRIS/PH5 #307: [BUG] pforma put wrong pickup time in array table for nodes

ascire-pic, 2019-06-19:
Pickup times generated by pforma to populate the array table for Fairfield node data were incorrect.

Used pforma to create a PH5 for a Fairfield node experiment. When I ran ph5_validate, it complained about finding 1k+ seconds of data after the pickup time for many stations (see orig-ph5_validate.log). 

After checking a few stations, I realized that the pickup times in the array table were incorrect. For example, for das 1X1 the last sample is at roughly Mar 19, 2019 21:18:55 (see 1x1_das-kef.txt) but the array table created by pforma has the pickup time Mar 19 2019 20:52:21 (see array1-fromPH5-20190613-kef.txt)

[orig-ph5_validate.log](https://github.com/PIC-IRIS/PH5/files/3306881/orig-ph5_validate.log)
[1x1_das-kef.txt](https://github.com/PIC-IRIS/PH5/files/3306883/1x1_das-kef.txt)
[array1-fromPH5-20190613-kef.txt](https://github.com/PIC-IRIS/PH5/files/3306884/array1-fromPH5-20190613-kef.txt)

derick-hess, 2019-06-20:
This is a known sort of bug. The nodes apparently continue to record after you disable acquisition and the node sets the internal pickup time. The pickup time is being read directly from the .fcnt files pickup time field. You may look at the data it says is after pickup time and see if it usable data. My guess is it probably isn't. 

This is why we made it just a warning in ph5_validate. There can be times when you want the pickup time to be before the end of the data because you know the data after pickup time is garbage. This could happen and be worse in cases where someone forgot to stop acquisition on the nodes and they got packed up and shipped across the country pack to the PIC while still recording. In that case pickup times would need to be set manually so PH5 has the option of not grabbing that data.

By default PH5 uses deploy pickup times for outputting data, but there is a flag to disable it to grab data that exists before and after pickup time.

If you wold like the nodes to always set the pickup time to be at the end of data that can be fixed in code. Instead of reading the pickup time stated in the file, the pickup time written in the array tables can just be calculated as one sample length after the last sample.

ascire-pic, 2019-06-20:
Hmm, okay I see what you mean. I went and pulled out some data from that ~30 minutes after the original pickup time pforma set for station 1 and it's just a flat line. I'm fine with leaving it as it is now since it makes sense to set the pickup time as the end of useable data. I'll make some modifications to our documentation to explain this so anyone using PH5 for nodes doesn't get confused when they look at the output from ph5_validate.

closed 2019-06-20 by ascire-pic