PIC-IRIS/PH5 #312: [BUG] Node redirected file names not working

dthomas1953, 2019-07-01:
**Describe the bug**
In previous versions of PH5, one would run unsimpleton to make the folder with the redirect file names for nodes (.rg16 files point to .fcnt files), and then, with resp_load, the ZLAND3C response file from NRL would be included in input.csv.
With this new version, the .rg16 file aliases don't seem to be working. On running resp_load, no ZLAND3C files were names in input.csv.

**Environment (please complete the following information):**
 - OS: OSX Yosemite 10.10.5
 - Program Version :ph5toms 2019.93

**To Reproduce**
Run ph5 with node data as usual, inspect the output of resp_load when you get to that point. It won't contain rows with the Node response file.

**Expected behavior**
I expected that input.csv would have rows naming the ZLAND3C response files.

**Additional context**
George Egert came up with a workaround using seg2toph5.
In the Sigma folder, run commands like this one, for single RAW node data files (.fcnt, not the .rg16 redirect):
seg2toph5 -n master.ph5 -r ../../NODE-raw/10.fcnt 
 
Then, running resp_load will include rows for the node response file(s).

dthomas1953, 2019-07-02:
I should add this happened on a project with three data sources, RT130s, Texans, and Nodes. I ran resp_load on a node-only dataset, and it added the ZLAND3C row to input.csv as expected. Will try a node-only run from scratch to see if that's OK too.

derick-hess, 2019-07-17:
Just in case. Make sure with multiple data sources you follow step 8 of the nutshell guide.

closed 2020-02-20 by nick-iris