PIC-IRIS/PH5 #313: [BUG] PH5 Install has begun failing...

dthomas1953, 2019-07-15:
**Describe the bug**
I've tried to install the latest PH5 on some Dell laptops and have consistent problems.

**Environment (please complete the following information):**
 - OS: CentOS Linux release 7.6.1810
 - Program Version [ph5toms 2019.93]

**To Reproduce**
Steps to reproduce the behavior:
Follow the steps for "Installing PH5 from source"
at https://github.com/PIC-IRIS/PH5/wiki/Installation-Instructions

**Expected behavior**
I expected PH5 to be installed, but that was not to be.

**Snippets**
This is from the install run.
```
[field@dhcp-10-10-10-41 ~]$ conda env list
# conda environments:
#
base                  *  /export/home/field/miniconda
nexus                    /export/home/field/miniconda/envs/nexus
passcal                  /export/home/field/miniconda/envs/passcal
ph5                      /export/home/field/miniconda/envs/ph5

[field@dhcp-10-10-10-41 ~]$ ls
2018148.92B7.00.ZIP  environment.yml                  passoft         Templates
2018148.92B7.ZIP     miniconda                        PH5             Videos
Desktop              MSEED                            Pictures
Documents            Music                            Public
Downloads            nexus-2018.247.c-py36_0.tar.bz2  setlogpeek.set
[field@dhcp-10-10-10-41 ~]$ cd PH5/
[field@dhcp-10-10-10-41 PH5]$ ls
CHANGELOG.txt    CONTRIBUTORS.txt  environment.yml  ph5        runtests.py
CONTRIBUTING.md  create_ph5.sh     LICENSE.txt      README.md  setup.py
[field@dhcp-10-10-10-41 PH5]$ conda env update --name=ph5 -f=environment.yml
Warning: you have pip-installed dependencies in your environment file, but you do not list pip itself as one of your conda dependencies.  Conda may not use the correct pip to install your packages, and they may end up in the wrong place.  Please add an explicit pip dependency.  I'm adding one for you, but still nagging you.
Collecting package metadata (repodata.json): done
Solving environment: **failed**

**_UnsatisfiableError: The following specifications were found to be incompatible with a past
explicit spec that is not an explicit spec in this operation (python):

  - numexpr -> numpy[version='>=1.9.3,<2.0a0'] -> python[version='>=2.7,<2.8.0a0']_**

The following specifications were found to be incompatible with each other:

  - basemap -> geos[version='>=3.7.2,<3.7.3.0a0'] -> libgcc-ng[version='>=7.3.0']
  - basemap -> geos[version='>=3.7.2,<3.7.3.0a0'] -> libstdcxx-ng[version='>=7.3.0']
  - basemap -> matplotlib-base[version='>=1.0.0'] -> backports.functools_lru_cache
  - basemap -> matplotlib-base[version='>=1.0.0'] -> cycler[version='>=0.10'] -> python -> bzip2[version='>=1.0.6,<2.0a0']
  - basemap -> matplotlib-base[version='>=1.0.0'] -> cycler[version='>=0.10'] -> python -> libffi[version='>=3.2.1,<3.3.0a0']
  - basemap -> matplotlib-base[version='>=1.0.0'] -> cycler[version='>=0.10'] -> python -> ncurses[version='>=6.1,<6.2.0a0']
  - basemap -> matplotlib-base[version='>=1.0.0'] -> cycler[version='>=0.10'] -> python -> openssl[version='>=1.1.1a,<1.1.2a'] -> ca-certificates
  - basemap -> matplotlib-base[version='>=1.0.0'] -> cycler[version='>=0.10'] -> python -> pip -> setuptools -> certifi[version='>=2016.09']
  - basemap -> matplotlib-base[version='>=1.0.0'] -> cycler[version='>=0.10'] -> python -> pip -> wheel
  - basemap -> matplotlib-base[version='>=1.0.0'] -> cycler[version='>=0.10'] -> python -> readline[version='>=8.0,<9.0a0']
  - basemap -> matplotlib-base[version='>=1.0.0'] -> cycler[version='>=0.10'] -> python -> sqlite[version='>=3.28.0,<4.0a0'] -> libedit[version='>=3.1.20181209,<3.2.0a0']
  - basemap -> matplotlib-base[version='>=1.0.0'] -> cycler[version='>=0.10'] -> python -> tk[version='>=8.6.9,<8.7.0a0'] -> zlib[version='>=1.2.11,<1.3.0a0']
  - basemap -> matplotlib-base[version='>=1.0.0'] -> cycler[version='>=0.10'] -> python -> xz[version='>=5.2.4,<5.3.0a0']
  - basemap -> matplotlib-base[version='>=1.0.0'] -> cycler[version='>=0.10'] -> six
  - basemap -> matplotlib-base[version='>=1.0.0'] -> freetype[version='>=2.9.1,<3.0a0'] -> libpng[version='>=1.6.35,<1.7.0a0']
  - basemap -> matplotlib-base[version='>=1.0.0'] -> functools32
  - basemap -> matplotlib-base[version='>=1.0.0'] -> icu[version='>=58.2,<59.0a0']
  - basemap -> matplotlib-base[version='>=1.0.0'] -> kiwisolver
  - basemap -> matplotlib-base[version='>=1.0.0'] -> numpy[version='>=1.14.6,<2.0a0'] -> libblas[version='>=3.8.0,<4.0a0'] -> libcblas==3.8.0=10_mkl -> liblapack==3.8.0=10_mkl -> liblapacke==3.8.0=10_mkl -> blas=[build=mkl] -> libgfortran-ng[version='>=7,<8.0a0']
  - basemap -> matplotlib-base[version='>=1.0.0'] -> numpy[version='>=1.14.6,<2.0a0'] -> libblas[version='>=3.8.0,<4.0a0'] -> mkl[version='>=2019.0,<2020.0a0'] -> intel-openmp
  - basemap -> matplotlib-base[version='>=1.0.0'] -> pyparsing
  - basemap -> matplotlib-base[version='>=1.0.0'] -> python-dateutil
[... 129 lines cut ...]
  - vispy -> numpy[version='>=1.14.6,<2.0a0'] -> libgfortran[version='>=3.0']
  - vispy -> numpy[version='>=1.14.6,<2.0a0'] -> numpy-base==1.16.4=py36hde5b4d6_0

[field@dhcp-10-10-10-41 PH5]$ source activate ph5

(ph5) [field@dhcp-10-10-10-41 PH5]$ ls
CHANGELOG.txt    CONTRIBUTORS.txt  environment.yml  ph5        runtests.py
CONTRIBUTING.md  create_ph5.sh     LICENSE.txt      README.md  setup.py

(ph5) [field@dhcp-10-10-10-41 PH5]$ pip install -e .
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Obtaining file:///export/home/field/PH5
    **ERROR: Complete output from command python setup.py egg_info:
    ERROR: Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/export/home/field/PH5/setup.py", line 10, in <module>
        from ph5.core.c_dependencies import subcd_py
      File "ph5/core/c_dependencies/subcd_py.py", line 8, in <module>
        raise ImportError(msg)
    ImportError: No module named numpy. Please install numpy first, it is needed before installing PH5.**
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in /export/home/field/PH5/
(ph5) [field@dhcp-10-10-10-41 PH5]$

```

dthomas1953, 2019-07-15:
Looks like numpy is not satisfied with Python 2.7 any longer...

nick-iris, 2019-07-15:
I just tried a fresh install of ph5 on my laptop and the install worked. You are trying to update an existing environment called `ph5`, so I think the conflict might be happening there. Please try creating a new environment (e.g. `ph5test`) and see if you are able to reproduce the problem.

dthomas1953, 2019-07-15:
OK, will do.

dthomas1953, 2019-07-15:
Tried that, got the same suite of errors. I will re-image the laptop in the morning, and start off from scratch. Might be something in the updated PASSOFT install.

nick-iris, 2019-07-15:
This is also causing Travis to fail for some pull requests. See https://travis-ci.org/PIC-IRIS/PH5/builds/556472184. Looks to be a real issue.

dthomas1953, 2019-07-15:
The messages were a little different, obspy was implicatyed as well.

```
[field@dhcp-10-10-11-249 PH5]$ conda env update --name=ph5test -f=environment.yml
Warning: you have pip-installed dependencies in your environment file, but you do not list pip itself as one of your conda dependencies.  Conda may not use the correct pip to install your packages, and they may end up in the wrong place.  Please add an explicit pip dependency.  I'm adding one for you, but still nagging you.
Collecting package metadata (repodata.json): done
Solving environment: failed

UnsatisfiableError: The following specifications were found to be incompatible with a past
explicit spec that is not an explicit spec in this operation (python):

  - obspy -> basemap -> numpy[version='>=1.14.6,<2.0a0'] -> python[version='>=2.7,<2.8.0a0']
```

and...
```
(ph5test) [field@dhcp-10-10-11-249 PH5]$ pip install -e .
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Obtaining file:///export/home/field/PH5
    ERROR: Complete output from command python setup.py egg_info:
    ERROR: Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/export/home/field/PH5/setup.py", line 10, in <module>
        from ph5.core.c_dependencies import subcd_py
      File "ph5/core/c_dependencies/subcd_py.py", line 8, in <module>
        raise ImportError(msg)
    ImportError: No module named numpy. Please install numpy first, it is needed before installing PH5.
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in /export/home/field/PH5/
(ph5test) [field@dhcp-10-10-11-249 PH5]$
```

derick-hess, 2019-07-16:
I saw this error on another project I'm working on. I changed around to appending conda-forge instead of adding it to lower it's priority (you can also set priority of conda-forge), then I just played around with pinning scipy, numpy and matplotlib to get it fixed. I think this is something happening in the conda and conda-forge channels.

dthomas1953, 2019-07-16:
I re-imaged the laptop from scratch, got the same errors upon installing PH5. @derick-hess, were you able to resolve this? How did you pin scipy/numpy etc.?

derick-hess, 2019-07-16:
figure out what version is needed and set that in the environment.yml file

dthomas1953, 2019-07-16:
OK, will give that a go

dthomas1953, 2019-07-17:
Meanwhile, on a Mac (OS Sierra 10.12.6), the install worked fine. This appears to be just an issue for the Linux Dells we're working with.

derick-hess, 2019-07-17:
I would start with scipy. That seemed to be causing most of the problems and from there this list shortened and I was able to change a few more versions and get everything working

derick-hess, 2019-07-17:
Oh and obspy is also having this issue as well. It seems harder to fix in python 2.7 than in python 3.

nick-iris, 2019-07-17:
I am unable to reproduce this issue on my mac. What version of conda and pip are you using?

I'm using the following:
```
pip 19.1.1 from /Users/nick/anaconda/envs/ph5test/lib/python2.7/site-packages/pip (python 2.7)
conda 4.6.14
```

dthomas1953, 2019-07-17:
@nick-iris , as I mentioned, it runs fine on my Mac also. It's just on our Linux boxes for the field (Dell Latitude 3490) that we are having the install issue. We are using CentOS Linux 7.6.1810, conda 4.7.5, and pip 19.1.1. I am trying to modify the environment.yml as @derick-hess suggested.

nick-iris, 2019-07-17:
The Travis CI test are also failing so we need to fix it before we can merge any new changes. I'll work on it as well.

dthomas1953, 2019-07-17:
I wonder if this has anything to due with Issue #312. It's looking like the resp_load issues might be due to a change in Python, perhaps something related to the deprecation of Python 2.7 for 2020. A few confirmations of problems so far.

dthomas1953, 2019-07-17:
Not getting very far with the environment.yml file. Here's what I tried:
```
dependencies:
 - hdf5=1.8.18
 - six
 - Cython
 - nose
 - numpy=1.16.4
 - scipy=1.3.0
 - numexpr
 - pyproj
 - PySide
 - pyqt=4
 - psutil
 - matplotlib<2
 - basemap
 - obspy
 - vispy
 - icu=56.*
 - libxslt=1.1.29=2
 - pip:
   - construct==2.5.1
   - simplekml
   - tables==3.4.2
   - flake8
   - pyflakes
   - coverage
```

And this is the error:
```
[field@dhcp-10-10-11-249 PH5]$ conda env update --name=ph5 -f environment.yml 
Warning: you have pip-installed dependencies in your environment file, but you do not list pip itself as one of your conda dependencies.  Conda may not use the correct pip to install your packages, and they may end up in the wrong place.  Please add an explicit pip dependency.  I'm adding one for you, but still nagging you.
Collecting package metadata (repodata.json): done
Solving environment: failed

UnsatisfiableError: The following specifications were found to be incompatible with a past
explicit spec that is not an explicit spec in this operation (python):

  - pip -> python[version='>=3.7,<3.8.0a0']

```

I tried pinning pip to 19.1.1, but it wasn't allowed.  Will try installing numpy and scipy overtly next.

dthomas1953, 2019-07-17:
Curious. On the CentOS Linux Dell, with no conda environment activated, python is version 3.6.7, and you can import scipy and numpy and see their version #s (1.3.0 and 1.16.4 respectively).

But, if you create a python 2.7 environment (similar to ph5's), and run python there (v. 2.7.15), you cannot import scipy and numpy from a python shell. In this 2.7 environment, I pip installed numpy and scipy, and now they can be imported into a python 2.7 shell (versions  1.16.4 and 1.2.2 respectively). 

However, trying the update to the ph5 environment using the .yml file, I got the same error, 
```
UnsatisfiableError: The following specifications were found to be incompatible with a past
explicit spec that is not an explicit spec in this operation (python):

  - pip -> python[version='>=3.7,<3.8.0a0']
```

Where is this demand for Python 3.7.x coming from? How do I find this "past explicit spec'?

nick-iris, 2019-07-17:
Trying adding an explicit pip dependency.

i.e. add ` - pip=19.1.1` above the `- pip:` line in the environment.yml file.

I get the following warning:

> Warning: you have pip-installed dependencies in your environment file, but you do not list pip itself as one of your conda dependencies.  Conda may not use the correct pip to install your packages, [...]

nick-iris, 2019-07-17:
Also try adding `- python=2.7` to the top of your dependencies list in the environment.yml file.

dthomas1953, 2019-07-17:
Tried both...
```
field@dhcp-10-10-11-249 PH5]$ cat environment.yml 
dependencies:
 - python=2.7
 - hdf5=1.8.18
 - six
 - Cython
 - nose
 - numpy=1.16.4
 - scipy=1.3.0
 - numexpr
 - pyproj
 - PySide
 - pyqt=4
 - psutil
 - matplotlib<2
 - basemap
 - obspy
 - vispy
 - icu=56.*
 - libxslt=1.1.29=2
 - pip=19.1.1
 - pip:
   - construct==2.5.1
   - simplekml
   - tables==3.4.2
   - flake8
   - pyflakes
   - coverage
```
... but I still get the same UnsatisfiableError. But, it was happy with the pip version, at least.

nick-iris, 2019-07-17:
Unfortunately I'm unable to reproduce this behavior. I even created a CentOS 7 virtual machine.

Try again with a fresh python environment. I'm wondering if you're running into issues because of versions that were left over from a previous install. I'd try removing the anaconda environment and installing again from the beginning.

nick-iris, 2019-07-17:
I'm going to create a new branch where I can push changes to the environment.yml to see if I can get TravisCI to pass.

dthomas1953, 2019-07-17:
I'll do a clean PH5 install in the morning, without any of our other passoft scripts, which may be related (those scripts have been updated recently).

nick-iris, 2019-07-17:
matplotlib is included with obspy so I think that dependency should be removed. Also I'm not sure if we need to explicitly include basemap.

nick-iris, 2019-07-17:
The problem is with conda version 4.7.5. Creating the environment works with conda version: 4.6.14. In the .travis.yml file we always grab the latest miniconda release. This is why this bug was introduced unexpectedly.

derick-hess, 2019-07-18:
also try this  while setting up conda. Anaconda changed around how their channels work, combining things, which has caused issues.

`conda config --set restore_free_channel true`

derick-hess, 2019-07-18:
A bit of context

> Why are things breaking, though? Symptoms include environments that are no longer satisfiable, or packages that are missing. All of the software on the “main” channel was carefully bootstrapped so that it does not include any reference to the free channel, but software from third party channels, such as conda-forge, may not have undergone such careful filtering for content from the free channel. Packages that end up depending on content that was only available on the free channel are now unsatisfiable. These packages need to be rebuilt. The hard part is often not rebuilding them, but identifying where the exact dependency on the free channel is introduced. We’re working on improving Conda’s messaging to help people understand this.

https://www.anaconda.com/why-we-removed-the-free-channel-in-conda-4-7/

derick-hess, 2019-07-18:
The travis.yml wont do anything except in github.

For local install just do the following on command line:

`conda config --set restore_free_channel true`

If that works then we can update the PH5 install instructions, also you will want to add that command to the PASSOFt and nexus installers as well so they dont break

dthomas1953, 2019-07-18:
I rebuilt the Dell Linux laptop, installed Anaconda 4.6.11, and the regular install ran fine, with no adjustments to environment.yml, and just using the Wiki install instructions.

I can try it with the new bells and whistles Derick mentioned as well.

dthomas1953, 2019-07-18:
I re-imaged the laptop using the new Passoft build (that's been causing the problems). This build has Conda installed, version 4.7.5. Installing PH5 as usul, I got the same error trying to update the ph5 environment.

Then, I updated the two .yml files as per above, and executed 
**conda config --set restore_free_channel true** 
before the env. update. This time, the update competed without error!

Now for the bad news: when I ran **pip install -e .**,   it did not compile ph5:
```
(ph5) [field@localhost PH5]$ pip install -e .
Obtaining file:///export/home/field/PH5
    ERROR: Complete output from command python setup.py egg_info:
    ERROR: Traceback (most recent call last):
      File "/export/home/field/PH5/setup.py", line 32, in <module>
        import PyQt4
    ModuleNotFoundError: No module named 'PyQt4'
    
    During handling of the above exception, another exception occurred:
    
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/export/home/field/PH5/setup.py", line 40, in <module>
        raise ImportError(msg)
    ImportError: No module named PyQt4. Please install PyQt4 first, it is needed before installing PH5.
    
    If using Anaconda run 'conda install pyqt=4'For pip users, PyQt4 installation instructions are available at http://pyqt.sourceforge.net/Docs/PyQt4/installation.html.
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in /export/home/field/PH5
```
My environment.yml file still has " - pyqt=4".

Suggestions?

dthomas1953, 2019-07-18:
I tried to install pyqt4 as per the error message, no luck:
```
(ph5) [field@localhost PH5]$ conda install pyqt=4
Collecting package metadata (current_repodata.json): done
Solving environment: done

# All requested packages already installed.

(ph5) [field@localhost PH5]$ pip install -e .
Obtaining file:///export/home/field/PH5
    ERROR: Complete output from command python setup.py egg_info:
. . .
```

dthomas1953, 2019-07-18:
Alas, the instructions to download sip went 404.
https://www.riverbankcomputing.com/static/Docs/sip4/installation.html

derick-hess, 2019-07-18:
install pyqt4 with apt-get on ubuntu.

sudo apt-get install python-qt4

dsentinel, 2019-07-18:
> install pyqt4 with apt-get on ubuntu.
> 
> sudo apt-get install python-qt4

I don't think we want to go this route.

dthomas1953, 2019-07-18:
**### SUCCESS**
Lloyd suggested I delete the ph5 environment, and re-do the install. 
This time, I got out of the (base) environment, re-created the ph5 environment with Python 2.7, executed the "conda config --set restore_free_channel true" command, and then ran the update for the ph5 env. And this time, the install (pip install -e .) worked fne, and PH5 is running, with the new laptop build, with Conda 4.7.5., in Linux.

dsentinel, 2019-07-18:
Good news... did the tests pass. can you run the utilities?

dthomas1953, 2019-07-18:
Results attached. Wondering if errors are related to Issue #312 ?
[ph5tests.txt](https://github.com/PIC-IRIS/PH5/files/3408630/ph5tests.txt)

derick-hess, 2019-07-18:
> Good news... did the tests pass. can you run the utilities?

If you look at #315 nick and I made the changes and all the CI runs and all tests pass

dthomas1953, 2019-07-23:
**FYI, just found out that restore_free_channel does not exist in Conda 4.6.**

Checking the new master install, I got this error:
`CondaValueError: Key 'restore_free_channel' is not a known primitive parameter`

I was using conda version : 4.6.8.

Then, I updated to Conda version : 4.7.10, and  restore_free_channel' can now be set without problems.

derick-hess, 2019-07-24:
Yes it only works in 4.7. 

4.6 uses free channel by default. free channel has been removed in 4.7 and above by default and will require the restore_free_channel option

dthomas1953, 2019-07-24:
Thanks, I updated the Wiki install page accordingly.

dsentinel, 2019-08-06:
#315 resolved this.
Closing.

closed 2019-08-06 by dsentinel