PIC-IRIS/PH5 #316: [BUG]

dthomas1953, 2019-07-24:
**Describe the bug**
There are errors in the test results. Some are actual python runtime errors. Others are warnings, some of which could be intentional (i.e. we were testing that the correct error messages are provided for incorrect inputs, etc.). I am attaching a copy of the test results for a run with the latest PH5 version.

**Environment (please complete the following information):**
 - OS: Sierra MacOS 10.12.6
 - Program Version entry_points.py: PROG_VERSION = '2019.162'

**To Reproduce**
After an install, execute:
```bash create_ph5.sh
python runtests.py
``` 

**Expected behavior**
No errors, warnings explained better. Lloyd Carothers thinks that there should not be output besides cumulating dots and numbers of tests run, and that only actual errors should be described fully in the output.

**Attachment**
The results are attached in the first comment.

dthomas1953, 2019-07-24:
Results are attached.
[runtests-7-24-2019.txt](https://github.com/PIC-IRIS/PH5/files/3428534/runtests-7-24-2019.txt)

derick-hess, 2019-07-24:
Looks like all the tests are passing to me. All of those errors it is catching are intentional and it is asserting they are happening.  You can modify the tests to not output the logging info to the screen. Not sure how extensive the modifications to the logging systems and tests would need to be.

Pretty much everything you see on the screen should be be from the logger output a few of those are generated by argparse, but everything is intentional.

You can see here all tests are passing and exit code 0 is sent notifying that all tests pass

https://travis-ci.org/PIC-IRIS/PH5/builds/562764559?utm_source=github_status&utm_medium=notification

I also suggest getting rid of runtests.py and just using the mose module to discover and run all tests. It is nicer and automatically will find all tests in the PH5 package and run them

derick-hess, 2019-07-24:
It may be as simple as catching the log and piping it to /dev/null.  All the assertions and tests should still work, just no logging to the screen

dthomas1953, 2019-07-24:
OK, thanks for the confirmation. I guess I just needed to see that the result file has intended errors. Will discuss piping options with the group.

dsentinel, 2019-07-24:
Erroneous output:
Tests should not print anything when run.
All the output done by the logging module can be cleanly reconfigured when running tests. Probably in setup.

runtests.py:
cleaned this up in #317 

for other print()'s:
these can be captured within a context manager... I think unittests has some or can be pathc mocked

dsentinel, 2019-07-24:
I am getting a stack of errors, many related to
```
APIError: (0, 'PH5 file does not exist: ph5/test_data/ph5/master.ph5')
```
```
$ ./runtests.py
EEEEEEEEEEEEEEEEEEEEEEE.EEEEEEEEEEEEEEEEEEEEE..........Successfully created the directory ph5/test_data/miniseedph5
[2019-07-24 20:54:20,998] - ph5.utilities.initialize_ph5 - INFO: Creating temporary receiver_t.tmp kef file using default values.
[2019-07-24 20:54:20,999] - ph5.utilities.initialize_ph5 - INFO: Loading Experiment_g/Receivers_g/Receiver_t using receiver_t.tmp.
usage: Version: 2019.63 metdatatoph5 --nickname="Master_PH5_file"-f "FILE" [options]
runtests.py: error: argument -n/--nickname is required
usage: Version: 2019.63 metdatatoph5 --nickname="Master_PH5_file"-f "FILE" [options]
runtests.py: error: argument -f/--file is required
usage: Version: 2019.63 metdatatoph5 --nickname="Master_PH5_file"-f "FILE" [options]
runtests.py: error: argument -n/--nickname is required
.Successfully created the directory ph5/test_data/miniseedph5
[2019-07-24 20:54:21,060] - ph5.utilities.initialize_ph5 - INFO: Creating temporary receiver_t.tmp kef file using default values.
[2019-07-24 20:54:21,060] - ph5.utilities.initialize_ph5 - INFO: Loading Experiment_g/Receivers_g/Receiver_t using receiver_t.tmp.
.Successfully created the directory ph5/test_data/miniseedph5
[2019-07-24 20:54:21,108] - ph5.utilities.initialize_ph5 - INFO: Creating temporary receiver_t.tmp kef file using default values.
[2019-07-24 20:54:21,109] - ph5.utilities.initialize_ph5 - INFO: Loading Experiment_g/Receivers_g/Receiver_t using receiver_t.tmp.
[2019-07-24 20:54:21,111] - ph5.utilities.metadatatoph5 - WARNING: ./master.ph5 not found. Creating...
[2019-07-24 20:54:21,149] - ph5.utilities.initialize_ph5 - INFO: Creating temporary receiver_t.tmp kef file using default values.
[2019-07-24 20:54:21,150] - ph5.utilities.initialize_ph5 - INFO: Loading Experiment_g/Receivers_g/Receiver_t using receiver_t.tmp.
[2019-07-24 20:54:21,151] - ph5.utilities.metadatatoph5 - INFO: Removing temporary receiver_t.tmp kef file.
[2019-07-24 20:54:21,157] - ph5.utilities.metadatatoph5 - INFO: Done... Created new PH5 file ./master.ph5.
[2019-07-24 20:54:21,218] - ph5.utilities.metadatatoph5 - INFO: File station.xml is STATIONXML...
[2019-07-24 20:54:21,218] - ph5.utilities.metadatatoph5 - INFO: *****************
[2019-07-24 20:54:21,218] - ph5.utilities.metadatatoph5 - INFO: Found station 0407
[2019-07-24 20:54:21,218] - ph5.utilities.metadatatoph5 - INFO: Found channel HHN
[2019-07-24 20:54:21,219] - ph5.utilities.metadatatoph5 - INFO: Response found for station 0407 channel HHN
[2019-07-24 20:54:21,219] - ph5.utilities.metadatatoph5 - INFO: Loading response for 0407 HHN
[2019-07-24 20:54:21,231] - ph5.utilities.metadatatoph5 - INFO: Response loaded
[2019-07-24 20:54:21,232] - ph5.utilities.metadatatoph5 - INFO: Loaded channel HHN
[2019-07-24 20:54:21,232] - ph5.utilities.metadatatoph5 - INFO: Found channel LHN
[2019-07-24 20:54:21,232] - ph5.utilities.metadatatoph5 - INFO: Response found for station 0407 channel LHN
[2019-07-24 20:54:21,232] - ph5.utilities.metadatatoph5 - INFO: Loading response for 0407 LHN
[2019-07-24 20:54:21,243] - ph5.utilities.metadatatoph5 - INFO: Response loaded
[2019-07-24 20:54:21,243] - ph5.utilities.metadatatoph5 - INFO: Loaded channel LHN
[2019-07-24 20:54:21,243] - ph5.utilities.metadatatoph5 - INFO: Loaded Station 0407
[2019-07-24 20:54:21,243] - ph5.utilities.metadatatoph5 - INFO: ******************

[2019-07-24 20:54:21,243] - ph5.utilities.metadatatoph5 - INFO: *****************
[2019-07-24 20:54:21,244] - ph5.utilities.metadatatoph5 - INFO: Found station 0407
[2019-07-24 20:54:21,244] - ph5.utilities.metadatatoph5 - INFO: Found channel LOG
[2019-07-24 20:54:21,244] - ph5.utilities.metadatatoph5 - INFO: Loaded channel LOG
[2019-07-24 20:54:21,244] - ph5.utilities.metadatatoph5 - INFO: Loaded Station 0407
[2019-07-24 20:54:21,244] - ph5.utilities.metadatatoph5 - INFO: ******************

.Successfully created the directory ph5/test_data/miniseedph5
[2019-07-24 20:54:21,415] - ph5.utilities.initialize_ph5 - INFO: Creating temporary receiver_t.tmp kef file using default values.
[2019-07-24 20:54:21,416] - ph5.utilities.initialize_ph5 - INFO: Loading Experiment_g/Receivers_g/Receiver_t using receiver_t.tmp.
[2019-07-24 20:54:21,431] - ph5.utilities.metadatatoph5 - INFO: File station.xml is STATIONXML...
[2019-07-24 20:54:21,431] - ph5.utilities.metadatatoph5 - INFO: *****************
[2019-07-24 20:54:21,432] - ph5.utilities.metadatatoph5 - INFO: Found station 0407
[2019-07-24 20:54:21,432] - ph5.utilities.metadatatoph5 - INFO: Found channel HHN
[2019-07-24 20:54:21,432] - ph5.utilities.metadatatoph5 - INFO: Response found for station 0407 channel HHN
[2019-07-24 20:54:21,432] - ph5.utilities.metadatatoph5 - INFO: Loading response for 0407 HHN
[2019-07-24 20:54:21,444] - ph5.utilities.metadatatoph5 - INFO: Response loaded
[2019-07-24 20:54:21,445] - ph5.utilities.metadatatoph5 - INFO: Loaded channel HHN
[2019-07-24 20:54:21,445] - ph5.utilities.metadatatoph5 - INFO: Found channel LHN
[2019-07-24 20:54:21,445] - ph5.utilities.metadatatoph5 - INFO: Response found for station 0407 channel LHN
[2019-07-24 20:54:21,445] - ph5.utilities.metadatatoph5 - INFO: Loading response for 0407 LHN
[2019-07-24 20:54:21,455] - ph5.utilities.metadatatoph5 - INFO: Response loaded
[2019-07-24 20:54:21,455] - ph5.utilities.metadatatoph5 - INFO: Loaded channel LHN
[... cut ...]
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/lloyd/code/PH5/ph5/core/tests/test_ph5api.py", line 13, in setUp
    nickname='master.ph5')
  File "/Users/lloyd/code/PH5/ph5/core/ph5api.py", line 221, in __init__
    os.path.join(path, nickname)))
APIError: (0, 'PH5 file does not exist: ph5/test_data/ph5/master.ph5')

----------------------------------------------------------------------
Ran 64 tests in 3.503s

FAILED (errors=44)
```

dthomas1953, 2019-07-24:
@dsentinel , did you execute `bash create_ph5.sh` before running the tests? That creates the necessary data structures.

dthomas1953, 2019-07-24:
Well, that's weird, I'm getting errors now, with the same setup as was working fine this morning. Will try a fresh install and test run.
```
----------------------------------------------------------------------
Ran 23 tests in 11.656s

FAILED (failures=11)
```

dthomas1953, 2019-07-24:
Re-ran a fresh build, and it ran 23 tests with an OK result. I will check out Lloyd's fix in branch fix_runtests to see if the test #s accumulate at the bottom for a clearer report.

dsentinel, 2019-07-24:
> @dsentinel , did you execute `bash create_ph5.sh` before running the tests? That creates the necessary data structures.

This should be in the tests `setup` function... and also `[2019-07-24 20:54:21,111] - ph5.utilities.metadatatoph5 - WARNING: ./master.ph5 not found. Creating...` 
The output claims it's created.

derick-hess, 2019-07-25:
> This should be in the tests `setup` function... and also
>
> ```
> [2019-07-24 20:54:21,111] - ph5.utilities.metadatatoph5 - WARNING: ./master.ph5 not found. Creating...
> ```
>
> The output claims it's [...]

This should definitely be in the setups for the tests. Really I think unit tests should be created for  the xxxtoph5 programs now, then the create_ph5.sh shouldn't be needed. That was only created originally because other things were higher priority to test and needed a PH5 to be generated. 

Might be good to just include a pre built PH5 in the test data to test against along with all the raw data. 

I think the test data should be added to setup.py as well to make it easier to reference. Your failure of the tests is probably because you weren't running runtests.py from the base package directory. Either  adding test data to setup.py or changing the way data is found in the tests to be in relation to the test file using something like:

```python
data_path = os.path.join(os.path.dirname(__file__), 'rest_of_path_to_data')
```
Thoughts?

dthomas1953, 2019-07-25:
I got the run_fixtests repo, and ran tests on a Mac Sierra 10.12.6.

```
(ph5) spike:PH5 dthomas$ git status
On branch fix_runtests
Your branch is up-to-date with 'origin/fix_runtests'.
```

```
...
======================================================================
ERROR: ph5.clients.tests.test_ph5tostationxml (unittest.loader.ModuleImportFailure)
----------------------------------------------------------------------
ImportError: Failed to import test module: ph5.clients.tests.test_ph5tostationxml
Traceback (most recent call last):
  File "/Users/dthomas/miniconda/envs/ph5/lib/python2.7/unittest/loader.py", line 254, in _find_tests
    module = self._get_module_from_name(name)
  File "/Users/dthomas/miniconda/envs/ph5/lib/python2.7/unittest/loader.py", line 232, in _get_module_from_name
    __import__(name)
  File "/Users/dthomas/ph5/PH5/ph5/clients/tests/test_ph5tostationxml.py", line 5, in <module>
    from ph5.clients.ph5tostationxml import PH5toStationXML
ImportError: cannot import name PH5toStationXML

----------------------------------------------------------------------
Ran 64 tests in 21.991s

FAILED (errors=1)
Closing remaining open files:ph5/test_data/ph5/master.ph5...doneph5/test_data/ph5/master.ph5...doneph5/test_data/ph5/master.ph5...done
```

derick-hess, 2019-07-25:
Tests need to be updated after @nick-iris updated the ph5tostationxml in pr #303

nick-iris, 2019-07-25:
Sorry about that

derick-hess, 2019-07-25:
I think... not sure what happened there, but that's first thought.

derick-hess, 2019-07-29:
I just took a look  and that error is coming from discovery of test_ph5tostationxml.py

That file is old, before the any of the current unit tests and before a redo of ph5tostationxml.  PH5tostationxml just needs new proper unit tests to be written so it is covered.

closed 2020-02-20 by nick-iris