PIC-IRIS/PH5 #320: [BUG] segdtoph5 not creating array table for multiple deployments

ascire-pic, 2019-07-30:
**Describe the bug**
A clear and concise description of what the bug is.

**Environment (please complete the following information):**
 - OS: [e.g. Ubuntu 16.04]
 - Program Version [ph5toms 2018.268]

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

damhuonglan, 2019-08-02:
I checked with segdtoph5:
+ with 9050.0.0.fcnt, it created in array table data with deploy_time=Tue Dec 11 11:26:16 2018
+ with 9050.1.0.fcnt, it created in array table data with deploy_time=Fri Dec 28 18:28:44 2018
Although the deploy_time of the 2nd one is lightly different with your fix, I still think it is correct.
Nex step, I will check with pforma to see if anything wrong with that.

damhuonglan, 2019-08-02:
@ascire-pic: I just push a commit for segdtoph5 that consider deploy time different when adding data to array table. However, there is an error in pforma when reading the raw data files which you seem not jump into it. I will see you on Monday to see what version you are using.

damhuonglan, 2019-08-06:
So we should use files.rg16 which are more descriptive than files.fcnt. However when I run pforma with new segdtoph5, it creates double data for the fist station in the raw list file created.

damhuonglan, 2019-08-06:
the reason that pforma duplicate data for the first station is the master.ph5 is copied to Sigma before adding other data which includes the data of the first station as well.
The above commit is a fix for this bug.
I will close this issue for now. You can reopen and give more comment if you see the problem still persist.

derick-hess, 2019-08-06:
I haven't looked at the commits yet to see if this was fixed, but it looks like you just committed changes to a branch for a different pull request.

To fix issues, create a new branch based off master, make the changes, commit and push to new branch. After this open a new pull request.

[Hot to contribute](https://github.com/PIC-IRIS/PH5/blob/master/CONTRIBUTING.md)

damhuonglan, 2019-08-06:
Thank you Derick. I already committed these changes to a new branch and create pull request for it.

 I just tried to commit too many things to remove_sys_exit. Is this possible for me to go back to a certain commit of remove_sys_exit that I already removed sys_exit and global variables from files and then try to merge that and move all other changes to a new branch?

derick-hess, 2019-08-06:
Yes, you can do a `git revert` on the two commits on that branch then push and it should be back to where it was before.

ascire-pic, 2019-09-10:
Issue appears to have been resolved in #341 

Test PH5 archive created using same data where problem was discovered has correct array tables for multiple deployments at a single station.

PR #358 (2020-01-16): Arrays of multideployment
This PR works on segd2ph5.py on the following issues:

- Change logger to LOGGER for consistency with others.
- Correct 'station' to 'das' in write_arrays().
- Remove not used functions.
- Fix #320 in ARRAY_T add dtime in dict level to allow more than one deploytime's arrays to be added.

### Checklist
- [ ] This PR is not directly related to an existing issue (which has no PR yet).
- [ ] All tests pass.
- [ ] Any new or changed features have are documented.
- [ ] Changes have been added to `CHANGELOG.txt` .
- [ ] First time contributors have added your name to `CONTR [... cut]
commit 11433d70 (2020-01-16): Arrays of multideployment (#358)

* change logger to LOGGER in segd2ph5.py for consistency

* correct 'station' to 'das' in write_arrays() in segd2ph5.py

* clean code by removing trash comments, not used functions: correct_append_number(), print_container() station_cmp()

* fix #320 in ARRAY_T add dtime in dict level to allow more than one deploytime's arrays to be added

* segd2ph5's unittest for the fix to #320

* create setLogger() which remove handler to set new handler file for LOGGER because the tmp/ folder is created for each test and delete right after

* add comments

* add initiatal [... cut]
closed 2020-01-16 by dsentinel